openwisp/netjsonconfig #58: [openwrt] Drop "enabled" logic in OpenVpnRenderer

nemesisdesign, 2017-01-17:
Solve the issues explained by the comments marked as `TODO` in https://github.com/openwisp/netjsonconfig/commit/7a152a344333665bbc9217011418ae39e8a1af81.

jack17529, 2017-03-21:
@nemesisdesign ok sir I will solve them.

nemesisdesign, 2017-03-21:
@jack17529 I did not specify that this must be done in the `0.6.0` release, not now. It's a reminder for a minor cleanup. Thanks anyway.

jack17529, 2017-03-22:
@nemesisdesign  so should I not touch the code?

nemesisdesign, 2017-03-22:
@jack17529 no it's not needed. I'll take care of it. I suggest you to find ways to contribute that are more useful to us. Stuff I don't have time to do, finding new bugs. Triaging tickets. Whatever you can do to help out.

jack17529, 2017-03-22:
@nemesisdesign  roger that .

commit a4020798 (2017-05-25): [openwrt] Dropped obsolete code in OpenVpn backend #58

Closes #58
commit 9ce3b344 (2017-05-26): [openwrt] Dropped obsolete code in OpenVpn backend #58

Closes #58
closed 2017-05-26 by nemesisdesign