openwisp/netjsonconfig #59: [openwrt] Improve multiple ip address output

nemesisdesign, 2017-01-26:
Such NetJSON DeviceConfiguration having 1 ipv4 and 1 ipv6 for each interface:

```javascript
{
    "interfaces": [
        {
            "bridge_members": [
                "eth0.1"
            ],
            "type": "bridge",
            "name": "lan",
            "addresses": [
                {
                    "proto": "static",
                    "family": "ipv4",
                    "mask": 24,
                    "address": "10.40.0.11",
                    "gateway": "10.40.0.9"
                },
                {
                    "proto": "static",
                    "family": "ipv6",
                    "mask": 64,
                    "address": "2001:4c00:893b:fede::1"
                }
            ]
        },
        {
            "type": "ethernet",
            "name": "eth0.24",
            "network": "vlan24",
            "description": "NanoBeam AC verso DiegoApriliana",
            "addresses": [
                {
                    "proto": "static",
                    "family": "ipv4",
                    "mask": 16,
                    "address": "172.16.40.24"
                },
                {
                    "proto": "static",
                    "family": "ipv6",
                    "mask": 128,
                    "address": "2001:4c00:893b:1:40::24"
                }
            ]
        },
        {
            "type": "ethernet",
            "name": "eth0.31",
            "network": "vlan31",
            "description": "NanoBeam AC verso RDP",
            "addresses": [
                {
                    "proto": "static",
                    "family": "ipv4",
                    "mask": 16,
                    "address": "172.16.40.31"
                },
                {
                    "proto": "static",
                    "family": "ipv6",
                    "mask": 128,
                    "address": "2001:4c00:893b:1:40::31"
                }
            ]
        }
    ]
}
```

generates this UCI output, which is quite messy:

```
config interface 'lan'
	option ifname 'eth0.1'
	option ipaddr '10.40.0.11'
	option netmask '255.255.255.0'
	option type 'bridge'
	option proto 'static'
	option gateway '10.40.0.9'

config interface 'lan_2'
	option ifname 'br-lan'
	option ip6addr '2001:4c00:893b:fede::1/64'
	option proto 'static'

config interface 'vlan24'
	option description 'NanoBeam AC verso DiegoApriliana'
	option ifname 'eth0.24'
	option ipaddr '172.16.40.24'
	option netmask '255.255.0.0'
	option proto 'static'

config interface 'vlan24_2'
	option description 'NanoBeam AC verso DiegoApriliana'
	option ifname 'eth0.24'
	option ip6addr '2001:4c00:893b:1:40::24/128'
	option proto 'static'

config interface 'vlan31'
	option description 'NanoBeam AC verso RDP'
	option ifname 'eth0.31'
	option ipaddr '172.16.40.31'
	option netmask '255.255.0.0'
	option proto 'static'

config interface 'vlan31_2'
	option description 'NanoBeam AC verso RDP'
	option ifname 'eth0.31'
	option ip6addr '2001:4c00:893b:1:40::31/128'
	option proto 'static'
```

The following cleaner output would be preferable:

```
config interface 'lan'
	option ifname 'eth0.1'
	option ipaddr '10.40.0.11'
	option ip6addr '2001:4c00:893b:fede::1/64'
	option netmask '255.255.255.0'
	option type 'bridge'
	option proto 'static'
	option gateway '10.40.0.9'

config interface 'vlan24'
	option description 'NanoBeam AC verso DiegoApriliana'
	option ifname 'eth0.24'
	option ipaddr '172.16.40.24'
	option ip6addr '2001:4c00:893b:1:40::24/128'
	option netmask '255.255.0.0'
	option proto 'static'

config interface 'vlan31'
	option description 'NanoBeam AC verso RDP'
	option ifname 'eth0.31'
	option ipaddr '172.16.40.31'
	option ip6addr '2001:4c00:893b:1:40::31/128'
	option netmask '255.255.0.0'
	option proto 'static'
```

This issue has lower priority because it's a clean-up task, the generated configuration works anyway, even if ugly.

ritwickdsouza, 2017-03-31:
Working on it.

ritwickdsouza, 2017-03-31:
https://github.com/openwisp/netjsonconfig/blob/master/netjsonconfig/backends/openwrt/renderers.py#L98

Is there any other condition when counter becomes greater than 1 ?

nemesisdesign, 2017-03-31:
@ritwickdsouza sure there is:
https://github.com/openwisp/netjsonconfig/blob/master/tests/openwrt/test_network.py#L41-L93

ritwickdsouza, 2017-03-31:
So the expected output for that should be this ?
```
package network
config interface 'eth0_1'
    option auto '1'
    option ifname 'eth0.1'
    option ipaddr '192.168.1.1'
    option ip6addr 'fd87::1/128'
    option netmask '255.255.255.0'
    option proto 'static'
config interface 'eth0_1_2'
    option auto '1'
    option ifname 'eth0.1'
    option ipaddr '192.168.2.1'
    option ip6addr 'fd87::1/128'
    option netmask '255.255.255.0'
    option proto 'static'
```

ritwickdsouza, 2017-04-01:
And why did the `lan` interface overwrite `ifname` of lan_2 from `br-lan` to `eth0.1` ?

nemesisdesign, 2017-04-01:
@ritwickdsouza no, OpenWRT has a way to specify multiple ip addresses on a single `config interface` directive.

I need to test something before giving you precise directions on this one.

ritwickdsouza, 2017-04-01:
Alright. Let me know once your testing is done.

nemesisdesign, 2017-04-01:
@ritwickdsouza I suggest you to announce the result of your work on the mailing list (and on social media if you use any). In the announcement, you could ask the community to help us testing it before releasing it. It's not guaranteed that somebody may help, but you never know :-)

nemesisdesign, 2017-05-08:
testing gitter integration

nemesisdesign, 2017-05-08:
test IRC integration

nemesisdesign, 2017-05-08:
test IRC integration 2

nemesisdesign, 2017-05-29:
@ritwickdsouza I have been able to work on this issue. It was quite hard, in retrospective I believe it was a good choice to suggest you to work on something else.

You can see my solution in https://github.com/openwisp/netjsonconfig/commit/179659c37223dc2363e76365a4514b2877ef04ce.

commit 179659c3 (2017-05-29): [openwrt] Improved multiple ip address output #59

Improved output of OpenWRT/LEDE UCI configuration
when using multiple ip addresses.
Cyclomatic complexity has been decreased thanks to
the split of many instructions in separate methods.

I tried to avoid changing the output of the library
when using single ipv4 or ipv6 addresses: this should
reduce the possibility of introducing bugs in systems
that have been already deployed.

Closes #59
closed 2017-05-29 by nemesisdesign