openwisp/netjsonconfig #61: [openwrt] Missing zonename attribute

nemesisdesign, 2017-02-17:
The `OpenWrt` backend does not generate the `zonename` attribute.

ritwickdsouza, 2017-03-07:
What is `zonename `? Is it different from `timezone`? Could you give a sample code for generating the bug ?

nemesisdesign, 2017-03-08:
@ritwickdsouza look in the OpenWRT docs: https://wiki.openwrt.org/doc/uci/system; to replicate the bug write a test in which you assert for the presence of the `zonename` attribute. The value should be generated from timezone.

ritwickdsouza, 2017-03-08:
So whenever `timezone` attribute is defined. Both `timezone` and `zonename` must be rendered. Right ?

nemesisdesign, 2017-03-08:
Yes

ritwickdsouza, 2017-03-09:
Could you help point out where exactly the output generated and places where changes needs to be made ? I'm having some trouble understanding the codebase.

nemesisdesign, 2017-03-09:
@ritwickdsouza: https://github.com/openwisp/netjsonconfig/blob/master/netjsonconfig/backends/openwrt/renderers.py#L265-L277

nemesisdesign, 2017-03-09:
I suggest you to write a failing test first (Test Driven Development), you have to add it to this file:
https://github.com/openwisp/netjsonconfig/blob/master/tests/openwrt/test_system.py

See the documentation regarding how to run tests.

nemesisdesign, 2017-03-13:
Fixed by #65

closed 2017-03-13 by nemesisdesign