openwisp/netjsonconfig #68: Cannot build documentation on a fresh developer install

EdoPut, 2017-04-23:
## Steps to reproduce:

- clone repo
- make virtualenv and install requirements.txt
- `python setup.py install`
- `cd docs`
- `make html`

## Expected result

The sphinx tool proceed to generate documentation in html

## Actual result

The sphinx tool fails to generate documentation in html, leaves a traceback

```
(env) edoput@hypnotoad:~/repo/netjsonconfig/docs$ make html
sphinx-build -b html -d build/doctrees   source build/html
Running Sphinx v1.5.5

Exception occurred:
  File "conf.py", line 20, in <module>
ImportError: No module named netjsonconfig.version
The full traceback has been saved in /tmp/sphinx-err-Jys2A5.log, if you want to report the issue to the developers.
Please also report this if it was a user error, so that a better error message can be provided next time.
A bug report can be filed in the tracker at <https://github.com/sphinx-doc/sphinx/issues>. Thanks!
Makefile:55: recipe for target 'html' failed
make: *** [html] Error 1
```

nemesisdesign, 2017-04-24:
If you open the python shell and run:

```python
from netjsonconfig.version import VERSION, get_version
print(VERSION)
print(get_version())
```

Do you get the `ImportError` or not?

If you do, install netjsonconfig with `python setup.py develop` as suggested in [Installing git fork for contributing](http://netjsonconfig.openwisp.org/en/stable/general/setup.html#install-git-fork-for-contributing).

EdoPut, 2017-04-24:
the netjsonconfig package is installed and the module can be imported

```python3
Python 3.5.3 (default, Jan 19 2017, 14:11:04) 
[GCC 6.3.0 20170118] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from netjsonconfig.version import VERSION, get_version
>>> VERSION
(0, 5, 5, 'post1')
>>> get_version
<function get_version at 0x7f64d8ce4ae8>
>>> get_version()
'0.5.5 post1'
```

```python2
Python 2.7.13 (default, Jan 19 2017, 14:48:08) 
[GCC 6.3.0 20170118] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> from netjsonconfig import VERSION, get_version
>>> VERSION
(0, 5, 5, 'post1')
>>> get_version
<function get_version at 0x7f0490bee140>
>>> get_version()
'0.5.5 post1'
>>> 
```

While investigating the issue  on python2 I was not able to reproduce both on python 2 and 3

This 
```
virtualenv --python=python2 env2
source env2/bin/activate
pip install -r requirements.txt 
python setup.py install
cd docs/
pip install sphinx
make html
```

The corresponding procedure for python3 does not yield any error as it produces the documentation

I'm leaving the output of `pip list` for future reference

```
alabaster (0.7.10)
appdirs (1.4.3)
Babel (2.4.0)
docutils (0.13.1)
imagesize (0.7.1)
Jinja2 (2.9.6)
jsonschema (2.6.0)
MarkupSafe (1.0)
netjsonconfig (0.5.5.post1)
packaging (16.8)
pip (9.0.1)
Pygments (2.2.0)
pytz (2017.2)
requests (2.13.0)
setuptools (35.0.1)
six (1.10.0)
snowballstemmer (1.2.1)
Sphinx (1.5.5)
wheel (0.29.0)
```

closed 2017-04-24 by EdoPut

nemesisdesign, 2017-04-24:
Maybe you forgot to activate the virtualenv.

EdoPut, 2017-05-08:
Resolved, sphinx must be installed in the virtualenv, the system install cannot reach the module

nemesisdesign, 2017-05-08:
@EdoPut thx for closing the issue