openwisp/netjsonconfig #79: [GSoC17] Raspbian phase 2

nemesisdesign, 2017-05-31:
@ritwickdsouza 

- Generate /etc/resolv.conf using Jinja2
- Generate /etc/hostapd/hostapd.conf using Jinja2
- Generate tar.gz file to install on Raspbian
- Improve tests
- Improve documentation
- First release

nemesisdesign, 2017-07-21:
@ritwickdsouza how far are we from achieving this?

ritwickdsouza, 2017-07-21:
@nemesisdesign WPA2 Enterprise and some advanced wireless features and modes like multiple ssids, wds mode are remaining. All other basic configurations can be generated. 
Generating tar.gz file is also done. I will once have to test tar.gz files for more complex NetJSON Device Configurations.
Documentations needs to be updated after scripts can be generated.

nemesisdesign, 2017-07-21:
Generate methods are ready right?

Work to introduce a basic customized schema for the backend so that all extra required attributes are present in the schema and enforced during validation.

This way we can proceed with a first release of the backend, we will have to write in the documentation page that this backend is experimental and so on.

ritwickdsouza, 2017-08-27:
The backend can generate Raspbian config files for various basic configurations. For more details have a look at http://netjsonconfig.openwisp.org/en/raspbian/backends/raspbian.html.

The backend can also generate tar.gz files to be installed on Raspbian

closed 2017-08-29 by nemesisdesign