openwisp/netjsonconfig #81: JSON schema does not add default values to configuration dictionary

EdoPut, 2017-06-05:
Right now we have a lot of material specified in the schema for a backend but it is not used to fill the defaults for the user,  so what is the benefit of having so much precise schema if it is not doing something for us?

Obviously this is a rant/question/"what would we like to do question", so please chime in and let's sort it out

nemesisdesign, 2017-06-05:
The default values are used by UI generators, eg: in the admin interface of OpenWISP, or in the [netjsonconfig-editor.js](https://github.com/openwisp/netjsonconfig-editor.js) project on which @gastonche is working.

Maybe there's a way to get a python dictionary filled with the default values of the schema via the jsonschema library, but I'm not sure.

I understand that AirOS needs a full configuration in order to work, so you need to keep a default config somewhere, let's try to go deeper, what's exactly you need / want to do? Can you work around this need without introducing too much complexity to the library?

EdoPut, 2017-06-05:
>AirOS needs a full configuration in order to work, so you need to keep a default config somewhere

I have been doing the defaults in the converters but I have many 'KeyError' poppings out during testing that (maybe) could be sorted out at jsonschema level

> Can you work around this need without introducing too much complexity to the library?

I think that yes I can do this without too much work but I should really look and see if jsonchema can do this on our behalf

nemesisdesign, 2017-06-06:
I've found something relevant, see: https://python-jsonschema.readthedocs.io/en/latest/faq/#why-doesn-t-my-schema-that-has-a-default-property-actually-set-the-default-on-my-instance

EdoPut, 2017-06-19:
If you are still interested in breaking every tests we have in one commit here is what I have worked on https://github.com/EdoPut/netjsonconfig/tree/computer-do-the-thing

nemesisdesign, 2017-06-19:
Looks interesting. I'm a bit overloaded now, but at some point in the near future I want to go back to analyse what could possibly go wrong in changing the current implementation to adopt that change and make it work for all backends.