openwisp/netjsonconfig #96: Openwrt backend validation problem for property ip_rules/src

hscadore, 2017-08-28:
Trying to generate a firewall rule for Openwrt causes an error because the library requires the zone definition "option src" (and option dest) to be an IP where they are actually zones according to openwrt documentation (https://wiki.openwrt.org/doc/uci/firewall).

I'm using netjsonconfig version 0.5.5.

nemesisdesign, 2017-08-28:
Link to original discussion: https://groups.google.com/d/msg/openwisp/bHqaeCBBYhI/bywnzXfSCwAJ

nemesisdesign, 2017-08-29:
After investigating this for a while, I found out there's a misunderstanding:

you are trying to use the policy routing feature (`ip rule`) instead of the firewall feature (which is basically a wrapper to iptables), **but the firewall feature is not supported in netjsonconfig yet** (you can use the [advanced mode and use the custom syntax to achieve what you need](http://netjsonconfig.openwisp.org/en/stable/backends/openwrt.html#all-the-other-settings)).

If you take a look at the documentation for ip rules in `/etc/config/network` you will see that `src` and `dest` are expected to be "*subnet to match (CIDR notation)*".

That said, it's still not good that the system fails with an exception, so I will address that issue for now.

commit cadb36b4 (2020-07-13): [fix] Openwrt backend validation problem for property ip_rules/src #96

Fixed issue by defining a new format checker for CIDR notation.

Fixes #96
PR #160 (2020-07-15): [fix] Openwrt backend validation problem for property ip_rules/src #96
Fixed the issue by adding a pattern to match the valid src and dest notation.

Fixes #96
closed 2020-07-15 by nemesisdesign