openwisp/netjsonconfig #104: [backward-conversion] fails when parsing LEDE config.tgz

jdeluyck, 2018-03-02:
Running a LEDE-generated tgz file through the following function causes a backtrace:

```
from netjsonconfig import OpenWrt
test = OpenWrt(native=open("config.tgz"))
print(test.json(indent=4))

```

```
Traceback (most recent call last):
  File "./test.py", line 10, in <module>
    test = OpenWrt(native=open("config.tgz"))
  File "/opt/openwisp2/env/lib/python3.5/site-packages/netjsonconfig/backends/base/backend.py", line 47, in __init__
    self.parse(native)
  File "/opt/openwisp2/env/lib/python3.5/site-packages/netjsonconfig/backends/base/backend.py", line 280, in parse
    self.to_netjson()
  File "/opt/openwisp2/env/lib/python3.5/site-packages/netjsonconfig/backends/base/backend.py", line 293, in to_netjson
    value = converter.to_netjson()
  File "/opt/openwisp2/env/lib/python3.5/site-packages/netjsonconfig/backends/base/converter.py", line 108, in to_netjson
    result = self.to_netjson_loop(block, result, index + 1)
  File "/opt/openwisp2/env/lib/python3.5/site-packages/netjsonconfig/backends/openwrt/converters/switch.py", line 70, in to_netjson_loop
    self._switch_map[vlan['device']].setdefault('vlan', [])
KeyError: 'switch0'
```

Curiously enough, when selecting just the files in etc/config/ and running them through the parser, works fine.

Code: 
```
from netjsonconfig import OpenWrt

foo = ""
for f in ['dhcp','dropbear','firewall','luci','network','openwisp','rpcd','system','ubootenv','ucitrack','uhttpd','wireless']:
  foo += open('etc/config/' + f, 'r').read()

test = OpenWrt(native=foo)

print(test.json(indent=4))
```

nemesisdesign, 2018-03-02:
The contents of the tar.gz are important in replicating the issue. The stackrace indicates a problem in parsing the switch configuration, if you could supply a dummy configuration which causes the same issue it would be helpful.

jdeluyck, 2018-03-03:
[test.zip](https://github.com/openwisp/netjsonconfig/files/1777516/test.zip)

Zipped tgz file. Contains one config file only, etc/config/network. It parses fine on its own, it fails horribly (same backtrace) when using the tgz.

stealthflame, 2018-12-03:
@nemesisdesign for this task also I need to add test cases?

PR #149 (2020-04-11): [bug] Fix switch vlan ordering #104
I found that the issue #104 is because the order of the list items is causing problems, `switch_vlan` is defined and processed before `switch` which is causing `KeyError` as switch itself is not existing at the time `switch_vlan` is being processed.

This pull request solves the problem by reordering the intermediate list.

Closes #104

Please let me know if there is a better way to solve this issue!
commit 772cec12 (2020-04-11): [bug] Fix switch vlan ordering #104

Closes #104
closed 2020-04-11 by nemesisdesign