openwisp/netjsonconfig #114: [json-schema] Test with jsonschema lib 3.0 alpha

nemesisdesign, 2018-08-07:
We should test netjsonconfig with [jsonschema v3.0.0a1](https://github.com/Julian/jsonschema/releases/tag/v3.0.0a1).

dwang, 2018-12-09:
Overall, it does not look like `jsonschema v3.0.0.a3` has any breaking changes compared to `jsonschema v2.6.0` currently on PyPI. 

I tested the changes by removing the current `jsonschema` from the requirements.txt file and running `pip install git+git://github.com/Julian/jsonschema.git@v3.0.0a3` to install from the GitHub tag. **Tests passed and nothing abnormal is seen.**

I also tried testing `django-netjsonconfig` as openwisp/django-netjsonconfig#63 said the bug may have been caused by the `jsonschema` library. **Tests pass again, but the same bug exists on the latest version of `jsonschema`.**

[image]

I think this is due to a separate reason as justinrainbow/json-schema#129, states the regex used to check the hostname is `^[_a-z]+.([_a-z]+.?)+$`. It does not include numbers or hyphens, which is why the bug occurs.

[This](https://stackoverflow.com/questions/106179/regular-expression-to-match-dns-hostname-or-ip-address) StackOverflow post states that the original RFC 952 did not allow hostnames to start with a digit, but it also says a later specification allowed numbers in the hostname ([RFC 1123](https://tools.ietf.org/html/rfc1123)).

I will try to fix this bug next.

Is there anything else I should address?

dwang, 2018-12-09:
Actually, it looks like `jsonschema v3.0.0a3` should have fixed this issue. I looked into the `json-schema` code more, and it looks like the regex that the library is using is `^[A-Za-z0-9][A-Za-z0-9\.\-]{1,255}$` which does include numbers. I'm looking into this right now.

dwang, 2018-12-09:
Or maybe it is completely unrelated?
I may be missing something here, but I do not see where `jsonschema` is being used for the web UI.
From what I can tell, the validation is done from `django_netjsonconfig/static/django-netjsonconfig/js/lib/advanced-mode.js`
```js
	    nonHostChars = ['%', '/', '?', ';', '#'].concat(autoEscape),
	    hostEndingChars = ['/', '?', '#'],
	    hostnameMaxLen = 255,
	    hostnamePartPattern = /^[a-z0-9A-Z_-]{0,63}$/,
	    hostnamePartStart = /^([a-z0-9A-Z_-]{0,63})(.*)$/,
```
```js
	    // the host is the remaining to the left of the first non-host char
	    hostEnd = -1;
	    for (var i = 0; i < nonHostChars.length; i++) {
	      var hec = rest.indexOf(nonHostChars[i]);
	      if (hec !== -1 && (hostEnd === -1 || hec < hostEnd))
	        hostEnd = hec;
	    }
	    // if we still have not hit it, then the entire thing is a host.
	    if (hostEnd === -1)
	      hostEnd = rest.length;

	    this.host = rest.slice(0, hostEnd);
	    rest = rest.slice(hostEnd);

	    // pull out port.
	    this.parseHost();

	    // we've indicated that there is a hostname,
	    // so even if it's empty, it has to be present.
	    this.hostname = this.hostname || '';

	    // if hostname begins with [ and ends with ]
	    // assume that it's an IPv6 address.
	    var ipv6Hostname = this.hostname[0] === '[' &&
	        this.hostname[this.hostname.length - 1] === ']';

	    // validate a little.
	    if (!ipv6Hostname) {
	      var hostparts = this.hostname.split(/\./);
	      for (var i = 0, l = hostparts.length; i < l; i++) {
	        var part = hostparts[i];
	        if (!part) continue;
	        if (!part.match(hostnamePartPattern)) {
	          var newpart = '';
	          for (var j = 0, k = part.length; j < k; j++) {
	            if (part.charCodeAt(j) > 127) {
	              // we replace non-ASCII char with a temporary placeholder
	              // we need this to make sure size of hostname is not
	              // broken by replacing non-ASCII by nothing
	              newpart += 'x';
	            } else {
	              newpart += part[j];
	            }
	          }
	          // we test again with ASCII char only
	          if (!newpart.match(hostnamePartPattern)) {
	            var validParts = hostparts.slice(0, i);
	            var notHost = hostparts.slice(i + 1);
	            var bit = part.match(hostnamePartStart);
	            if (bit) {
	              validParts.push(bit[1]);
	              notHost.unshift(bit[2]);
	            }
	            if (notHost.length) {
	              rest = '/' + notHost.join('.') + rest;
	            }
	            this.hostname = validParts.join('.');
	            break;
	          }
	        }
	      }
	    }

	    if (this.hostname.length > hostnameMaxLen) {
	      this.hostname = '';
	    } else {
	      // hostnames are always lower case.
	      this.hostname = this.hostname.toLowerCase();
	    }
```

```js
	    if (it.opts.messages !== false) {
	      out += ' , message: \'should match format "';
	      if ($isData) {
	        out += '\' + ' + ($schemaValue) + ' + \'';
	      } else {
	        out += '' + (it.util.escapeQuotes($schema));
	      }
	      out += '"\' ';
	    }
```

nemesisdesign, 2018-12-09:
Yes you're right, it's because of the javascript jsonschema validator.
Thanks a lot for your report Daniel, it's very useful! I will leave this open to ensure this stays on my radar.

nemesisdesign, 2019-11-30:
Follow up on #136.

closed 2019-11-30 by nemesisdesign