openwisp/netjsonconfig #117: [openwrt] Allow configure DNS resolvers without the need of defining interfaces

nemesisdesign, 2018-09-30:
In OpenWRT, dns settings are configured at interface level, you need to define some interfaces to make this config effective on OpenWRT.

Eg:

```
{
    "interfaces": [
       { ...something here... }
    ]
    "dns_servers": [
        "8.8.8.8",
        "8.8.4.4"
    ]
}
```

But this is not very good. There should be a way to make the DNS configuration work also when `interfaces` are not defined.

NoumbissiValere, 2019-03-07:
I have gone through the codebase and found out that the dns converters are defined within the interface converters. so if there is no interface in the `netjson config dict` the interface converters won't be executed and consequently the dns converter. I think this issue can be fixed by defining the dns converter out of the interface converter but it should still be display under the network package.

NoumbissiValere, 2019-03-26:
I went through the openwrt documentation and it seems that dns configuration on openwrt can be done only in two places , i.e under the `DHCP and DNS` and `Interface` options under `Network`. under the `DHCP and DNS` option one can set a `DNS forwarding` which enables openwrt to send requests to a dns server for a particular domain name. but this is not the functionality raised by the issue. The second option will be to make use of the `Interface` option. 
https://openwrt.org/docs/guide-user/base-system/dns_configuration
https://openwrt.org/docs/guide-user/base system/dns_configuration#enabling_dns_without_enabling_dhcp .
With these, i don't think i have seen a possibility of making the dns settings as stated in the issue available on the openwrt device without using an interface. 
Please i will love to be corrected if am wrong so that i correct my mistake and learn.

Blackyfff, 2021-04-10:
As far as I understand, the option to use the `Interface` is the recommended one.

Setting it in `DHCP and DNS` is only a setting for the package dnsmasq. If dnsmasq is installed it will have an effect, because OpenWRT will set the resolv.conf to 127.0.0.1 and DNS lookups are forwarded to local dnsmasq.
But with 'dump' APs you may not need dnsmasq, and then OpenWRT will set the DNS-Servers from the interfaces in the `resolv.conf`, so this option has no effect.

Another option is to directly manipulate the `resolv.conf`, but that may undermine some OpenWRT mechanisms and may break other things, but can be done at own risk.