openwisp/netjsonconfig #118: TypeError: can only concatenate list (not "str") to list

imp1sh, 2018-10-13:
I made a backup of two different systems (Archer C7 v2 and X86_64 Qemu guest).
Both I tried to reverse with the following lines:

```
from netjsonconfig import OpenWrt
#router = OpenWrt(native=open('./backup-g0.sim.libcom.de-2018-10-10.tar.gz'))
router = OpenWrt(native=open('./backup-OpenWrt-2018-10-10.tar.gz'))
print(router.render)
```

In both cases this results in the following error:
python reverse.py 
```
Traceback (most recent call last):
  File "reverse.py", line 3, in <module>
    router = OpenWrt(native=open('./backup-OpenWrt-2018-10-10.tar.gz'))
  File "/usr/lib/python2.7/site-packages/netjsonconfig/backends/base/backend.py", line 47, in __init__
    self.parse(native)
  File "/usr/lib/python2.7/site-packages/netjsonconfig/backends/base/backend.py", line 280, in parse
    self.to_netjson()
  File "/usr/lib/python2.7/site-packages/netjsonconfig/backends/base/backend.py", line 293, in to_netjson
    value = converter.to_netjson()
  File "/usr/lib/python2.7/site-packages/netjsonconfig/backends/base/converter.py", line 108, in to_netjson
    result = self.to_netjson_loop(block, result, index + 1)
  File "/usr/lib/python2.7/site-packages/netjsonconfig/backends/openwrt/converters/interfaces.py", line 206, in to_netjson_loop
    interface = self.__netjson_interface(block)
  File "/usr/lib/python2.7/site-packages/netjsonconfig/backends/openwrt/converters/interfaces.py", line 217, in __netjson_interface
    interface = self.__netjson_addresses(interface)
  File "/usr/lib/python2.7/site-packages/netjsonconfig/backends/openwrt/converters/interfaces.py", line 262, in __netjson_addresses
    for address in ipv4 + ipv6:
TypeError: can only concatenate list (not "str") to list
```
[new.tar.gz](https://github.com/openwisp/netjsonconfig/files/2475301/new.tar.gz)

nemesisdesign, 2018-10-18:
This is somewhat related to #104, the backward conversion process needs to be improved.
I'm adding these two tasks to the task list for the [Google Code-In program](https://codein.withgoogle.com/).

nemesisdesign, 2018-11-09:
@imp1sh, @AnonGuy is doing great work to fix the issue you reported in #119. Thanks again for your report. I hope we'll get it fixed very soon.

nemesisdesign, 2018-11-09:
(actually his patch should already work, only some minor corrections are missing)

stealthflame, 2018-11-22:
If no one is working on this I have taken this task under GCI. Also, It would be great to get the current position of the bug. :slightly_smiling_face:

PR #124 (2018-11-27): GCI Task: [netjsonconfig] TypeError: can only concatenate list (not "str") to list
This Issue is in reference to the issue #118 . It would be great to have your views on the changes made.
Thank you in advance. 

Closes #118
closed 2018-11-27 by nemesisdesign