openwisp/netjsonconfig #132: [openvpn] format of remote host should allow both hostname and ipv4

okraits, 2019-07-18:
A remote host can be a hostname or an ip address. Right now only a hostname is valid:
https://github.com/openwisp/netjsonconfig/blob/master/netjsonconfig/backends/openvpn/schema.py#L445

I tried with:
```
                                    "host": {
                                        "type": "string",
                                        "maxLength": 63,
                                        "minLength": 1,
                                        "oneOf": [
                                            {"format": "hostname"},
                                            {"format": "ipv4"},
                                        ],
                                        "propertyOrder": 1,
                                    },
```
But that doesn't work.
@nemesisdesign Any idea?

nemesisdesign, 2019-07-18:
Annoying problem, maybe we can just remove the `format` validation since it doesn't seem to be doable in a simple manner.

okraits, 2019-07-19:
Or maybe better: use a pattern to check for the two possible format options.

nemesisdesign, 2019-11-29:
Yes, let's use a pattern.

nemesisdesign, 2019-12-13:
I wonder if the new version of jsonschema solves this. #139

PabloCastellano, 2020-05-07:
Did you trying `anyOf` instead of `oneOf`?

nemesisdesign, 2020-05-07:
@PabloCastellano I tried it and it works but it needs some CSS adjustment in openwisp-controller because the UI breaks. Thanks for the tip!

nemesisdesign, 2020-07-06:
So, the problem does not lie in this library.
Using an hostname works fine.
The problem manifests when using the advanced mode of openwisp-controller.

nemesisdesign, 2020-07-06:
PS: See https://github.com/openwisp/netjsonconfig/pull/158 for confirmation of my claim.

commit dfd7c789 (2020-07-06): [tests] Ensure ip is valid remote in OpenVPN backend #132

Closes #132
PR #158 (2020-07-06): [tests] Ensure ip is valid remote in OpenVPN backend #132
Closes #132
closed 2020-07-06 by nemesisdesign