openwisp/netjsonconfig #133: Python3 compatibility

imp1sh, 2019-08-04:
Hi,

when I install netjsonconfig via pip3 I cannot import in python. This is the error message:
```
>>> from netjsonconfig import OpenWrt
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/setup/src/netjsonconfig/netjsonconfig/__init__.py", line 6, in <module>
    from .backends.openwrt.openwrt import OpenWrt  # noqa
  File "/home/setup/src/netjsonconfig/netjsonconfig/backends/openwrt/openwrt.py", line 1, in <module>
    from . import converters
  File "/home/setup/src/netjsonconfig/netjsonconfig/backends/openwrt/converters/__init__.py", line 1, in <module>
    from .default import Default
  File "/home/setup/src/netjsonconfig/netjsonconfig/backends/openwrt/converters/default.py", line 1, in <module>
    import json
  File "/home/setup/json.py", line 1, in <module>
    from netjsonconfig import OpenWrt
ImportError: cannot import name 'OpenWrt' from 'netjsonconfig' (/home/setup/src/netjsonconfig/netjsonconfig/__init__.py)
```
When I  instead run netjsonconfig with python2 I don't get that error. I suspect it's due to lacking python3 support. Since general python2 supports ends 2020 what do you intend to do about it?

imp1sh, 2019-08-04:
Well I don't get it. Now it's giving me that error in both python2 and 3. What's going on?

nemesisdesign, 2019-11-29:
Hi @imp1sh, sorry for my late reply. Were you able to solve it? I haven't encountered this issue anywhere.

imp1sh, 2019-11-30:
I don't remember. Not using netjsonconfig any more right now. I think this should be closed.

closed 2019-11-30 by imp1sh