openwisp/netjsonconfig #134: [bug] Cert matching query does not exist

atb00ker, 2019-09-06:
When deleting a Cert created by a device which looks like:
```
Summary
    Certificates: 1
    VPN clients: 1
Objects
    Certificate: 08:00:27:25:34:89
        VPN client: VpnClient object (4)
```

Further testing needs to be done to reproduce the `query does not exist` error.

nemesisdesign, 2019-11-29:
Hey @atb00ker, it looks like this issue was created in the wrong repository!
Shouldn't this be opened in django-netjsonconfig?

atb00ker, 2019-12-02:
Unfortunely, I don't remember a lot about the issue. However, It looks like a django related issue. 
I cannot recall why I opened it here, maybe by mistake, so I am moving it to `django-netjsonconfig` for now.