openwisp/netjsonconfig #143: Documentation link is redirected

jonathanunderwood, 2019-12-25:
Clicking on the documentation link in the README briefly loads the documentation, and then the browser is redirected to:

http://ww38.nodeshot.org/github-btn.html?user=openwisp&repo=netjsonconfig&type=watch&count=true&size=large&_xafvr=MWI1MGI3MmEwMGRlNWQ2YjMyNzgyYTZjN2EwYzYzNDdmOWM2ZTJhOCw1ZTAzMmMzMTIyZTdl

IAmAMissingColon, 2019-12-25:
+1 Same here ! But I thought it was browser acting weird

atb00ker, 2019-12-25:
Thanks for report, this seems to be unintensional and some kind of misconfiguration.
Might be related to: https://github.com/openwisp/netjsonconfig/blob/ec56134759866919040db0015c3acb77180a5091/docs/source/_github.rst

But we need to investigate further to solve it.

IAmAMissingColon, 2020-01-09:
is this issue fixed ?

jonathanunderwood, 2020-01-09:
Yes, it appears to be fine now.

jonathanunderwood, 2020-01-09:
The question is: how did this happen in the first place? Subversively collecting geolocation data is not what I expect from an open source project.

nemesisdesign, 2020-01-09:
I updated the contents of the docs with the updated link: https://github.com/openwisp/netjsonconfig/commit/ddc63c3e8e0a6742baca4a7e188e46f89603e645.
But we need to issue a new release in order to have it updated on the generated docs.
The older docs may stay like that forever unless we delete them.

> The question is: how did this happen in the first place? Subversively collecting geolocation data is not what I expect from an open source project.

What are you talking about? I was the maintainer of another project which the new version of OpenWISP initially based some code on, which was called [nodeshot](https://github.com/ninuxorg/nodeshot), and I used the domain I had to serve some static content which displayed a star and watch on github button, then the domain expired.

You've been pretty fast to jump to negative conclusions.

jonathanunderwood, 2020-01-11:
Ok, apologies. Given the nature of nodeshot, I assumed it was some attempt to geo track.

closed 2020-01-11 by jonathanunderwood

nemesisdesign, 2020-01-14:
Update: I've updated http://netjsonconfig.openwisp.org/ to make sure the redirect has been fixed and I removed the older versions which have the redirect (because they point to the expired domain).