openwisp/netjsonconfig #147: Add DEFAULT_FILE_MODE as a property

TheOneAboveAllTitan, 2020-03-21:
Currently, `DEFAULT_FILE_MODE` is defined as a constant and file permissions while using `OpenVpn.auto_client` method defaults to 0644 even when schema defines it otherwise. In my opinion, it will be better if, `DEFAULT_FILE_MODE` gets its value from the schema and not from hard code.

Please have a look at f[ollowing code](https://github.com/openwisp/netjsonconfig/blob/75a77efdc9782b0d16b910b82f2d7f58951da232/netjsonconfig/backends/openvpn/openvpn.py#L113).

nemesisdesign, 2020-03-21:
Sounds overkill to solve openwisp/django-netjsonconfig#169.

Changing these lines of code is enough: https://github.com/openwisp/netjsonconfig/blob/85dde64e2ebde871044a79bda445d0a2729ec58a/netjsonconfig/backends/openvpn/openvpn.py#L124-L134

We can add a new constant, `X509_FILE_MODE = '0600'` and use that in place of `DEFAULT_FILE_MODE`.

Please do not add complexity unless it's really needed.

I'm renaming this issue so we can use it to track the change needed to solve openwisp/django-netjsonconfig#169.

TheOneAboveAllTitan, 2020-03-22:
I was aiming for reusability and ease in customization and extending the app. Something like this works and does not increase much complexity. 

```
class OpenVpn(BaseBackend):
    """
    OpenVPN 2.x Configuration Backend
    """
    schema = schema
    converters = [converters.OpenVpn]
    parser = OpenVpnParser
    renderer = OpenVpnRenderer
    list_identifiers = ['name']

    @staicmethod
    def _get_file_mode(cls):
        return cls.schema['properties']['files']['items']['properties']['mode']['default']`

    @classmethod
    def _auto_client_files(cls, client, ca_path=None, ca_contents=None, cert_path=None,
                           cert_contents=None, key_path=None, key_contents=None):
        """
        returns a list of NetJSON extra files for automatically generated clients
        produces side effects in ``client`` dictionary
        """
        files = []
        if ca_path and ca_contents:
            client['ca'] = ca_path
            files.append(dict(path=ca_path,
                              contents=ca_contents,
                              mode=cls._get_file_modegit co()))
        if cert_path and cert_contents:
            client['cert'] = cert_path
            files.append(dict(path=cert_path,
                              contents=cert_contents,
                              mode=cls._get_file_mode()))
        if key_path and key_contents:
            client['key'] = key_path
            files.append(dict(path=key_path,
                              contents=key_contents,
                              mode=cls._get_file_mode(),))
        return files

```

nemesisdesign, 2020-03-27:
@TheOneAboveAllTitan nice but it's not needed, because the file mode to use will be always the same ('0600`). Keep it simple please.

One extra line of complexity in all patches and in 1 year we'll end up with 1000 extra lines of unneeded complexity. Be careful.

PR #148 (2020-03-28): [x509] Added dedicated file mode for generated x509 certificates #147
OpenVPN certificate files will have `0600` default permission.
Modified tests for same.
Related https://github.com/openwisp/django-netjsonconfig/issues/169
Closes #147
commit 10e4e8d3 (2020-03-28): [x509] Added dedicated file mode for generated x509 certificates #147

OpenVPN certificate files will have `0600` default permission.
Modified tests for same.
Closes #147
closed 2020-03-28 by nemesisdesign