openwisp/netjsonconfig #150: [netjsonconfig-tests] Avoid error logging to print noisy output during tests

nemesisdesign, 2020-04-11:
**Reserved for Google Code-In 2020**.

Depends on https://github.com/openwisp/openwisp-utils/issues/98.

Use the decorators mentioned in the issue linked above to silence the output of noisy tests.

lehone-hp, 2021-01-03:
Hello, 
when I run the tests, I get the following output
```
....................................................................................................................................................................................................................................................................................................Unrecognized config block was skipped:

{
    "unrecognized": true
}

.....................................................
----------------------------------------------------------------------
Ran 345 tests in 10.715s

OK

```
Please how can I know the exact test that runs and generates the output, `Unrecognized config block was skipped` for example like an additional parameter that I can append to the `./runtests.py`

2001asjad, 2021-01-07:
@lehone-hp  I wanted to ask how exactly did you come to know that what file in the tests folder we have to use the decorators in ?

lehone-hp, 2021-01-08:
> @lehone-hp I wanted to ask how exactly did you come to know that what file in the tests folder we have to use the decorators in ?

I ran the tests with the `-v` flag; `./runtests.py -v`, it lists all the tests methods as they run.

2001asjad, 2021-01-09:
> > @lehone-hp I wanted to ask how exactly did you come to know that what file in the tests folder we have to use the decorators in ?
> 
> I ran the tests with the `-v` flag; `./runtests.py -v`, it lists all the tests methods as they run.

@lehone-hp but when I run `./runtests.py -v` command for solving issue [https://github.com/openwisp/openwisp-controller/issues/187](url) to know which test methods are giving noisy outputs I get this error :

[image]
Can you help me out in this ?
`./runtests.py` shows this : 

[image]

Thank You

lehone-hp, 2021-01-09:
@2001asjad try, `python runtests.py -v 2`

2001asjad, 2021-01-09:
> @2001asjad try, `python runtests.py -v 2`

It works thank you so very much @lehone-hp 😊

PR #176 (2021-01-11): [netjsonconfig-tests] Avoid error logging to print noisy output durin…
…g tests #150

Applied `capture_stdout` decorator on noisy tests

Fixes #150
commit 3d53e149 (2021-01-11): [tests] Avoided noisy output during tests #150

Applied `capture_stdout` decorator on noisy tests.

Closes #150
closed 2021-01-11 by nemesisdesign