openwisp/netjsonconfig #153: [config] Merging lists

nemesisdesign, 2020-05-02:
The way configuration is merged with templates now has a quirk:
list elements from different templates are merged with the config.

This can be useful, but it's weird.

I wonder if it's possible to support both ways at the same time, letting the user decide which one to use in the advanced mode.

EG:

- for this particular configuration, I want the list value present in the device config to override the list in the template
- for this other configuraiton, I have a long list in the template (or 2 templates with 2 lists that are merged) and I only want the device configuration to list the additional item, but I do not want to redefine the whole list

The question is how? This would be great to have. It would have to be documented properly.

okraits, 2023-06-16:
Let's list the options we have so we can get closer to a solution.

1. Items of lists with the same name in different templates are merged, resulting in a list containing all items from all templates. This is the current behaviour.
2. If there are lists with the same name in different templates, the items of the list in the template with the highest priority will be the items in the resulting list, overriding the items in the other templates. This has the disadvantage that for adding items to a list, all common items need to be listed in several templates.
3. We could add an option/setting to the list in our templates which defines the behaviour of that list if there is a list with the same name in a template with lower priority: add items to that lower priority list or replace items. This makes it possible to choose the behaviour depending on the usecase.

@nemesisdesign  What do you think?

nemesisdesign, 2023-06-16:
@okraits option 3. sounds like a good idea, an "advanced option" which allows to decide if the user wants to add to any previous template or override, to be set on a per template basis and it would act only on previous templates, right?

okraits, 2023-06-19:
> @okraits option 3. sounds like a good idea, an "advanced option" which allows to decide if the user wants to add to any previous template or override, to be set on a per template basis and it would act only on previous templates, right?

Yes, each template would basically take the result of the previous templates as input and then override that or add to it, based on the selected advanced option.

okraits, 2023-09-14:
@nemesifier So if you're ok with option 3, I will start to work on it, ok?

nemesifier, 2023-09-14:
@okraits how do you intend to implement this? Are we going to add an initialization argument to the library?

You mention adding an "option" to the list, can you expand on this?

okraits, 2023-09-14:
> @okraits how do you intend to implement this? Are we going to add an initialization argument to the library?
> 
> You mention adding an "option" to the list, can you expand on this?

@nemesifier 

To be honest, I have to look into the code first to see what's possible. I'll get back to you with a proposal, ok?

okraits, 2023-10-16:
@nemesifier I thought about the whole topic and possible solutions.

Originally I thought about adding a "meta setting" to every list in the json configuration dictionary (which would mean that we would have to store a list in json in a surrounding dictionary to be able to have a "meta setting") to define wether the corresponding list should be merged with or override an existing list.
But I guess that this is too complex, we would need to extend the json schema and possibly modify all the parsers, converters and renderers.

So I guess the more simple solution is to add a field to the _AbstractTemplate_ model to define the behaviour of lists for each template. I can think of the following behaviours:

* override/replace an existing list
* insert items at the beginning of an existing list
* append items to the end of an existing list

Am I correct that we basically need to modify the following functions to implement the behaviours mentioned above:

* https://github.com/openwisp/netjsonconfig/blob/master/netjsonconfig/utils.py#L6
* https://github.com/openwisp/netjsonconfig/blob/master/netjsonconfig/utils.py#L35

What do you think?

okraits, 2023-10-23:
@nemesifier Any opinion on my suggestion?

okraits, 2023-11-09:
@nemesifier I didn't receive a response from you, so I implemented this. Please take a look at it and feel free to ask questions or request changes.

commit 36700136 (2023-11-09): [feature] Add possibility to change list handling behaviour #153

Closes #153
commit 2a05ab47 (2023-11-09): [feature] Add possibility to change list handling behaviour #153

Closes #153