openwisp/netjsonconfig #166: [bug] File path shouldn't be allowed to be empty

nemesisdesign, 2020-10-14:
I am not sure if this is intentional or not, but the file path shouldn't be empty.

balajiprabuk, 2020-10-19:
Can you please provide details about which file path you are referrring to. Need few more details about this issue.

nemesisdesign, 2020-10-19:
@balajiprabuk: thanks for requesting more clarification, I meant here: https://github.com/openwisp/netjsonconfig/blob/master/netjsonconfig/schema.py#L911

It looks like `path` is required but in one case I was able to save a NetJSON document which had a file in which the path was not specified. I think I should try to collect more information before we'll be able to find out where the issue lies.

2001asjad, 2021-01-08:
@nemesisdesign Hello, did you find more about this issue so that we can work on it ?
Thanks

nemesisdesign, 2022-01-31:
I think it's possible to save a file which has an empty path, which is wrong, I think we should require the path to be at least 2 characters long.

PR #218 (2022-03-10): [Fix] File path shouldn't be allowed to be empty #166
Fixes #166
commit 109b3f42 (2022-03-10): [fix] File path shouldn't be allowed to be empty #166

Fixes #166