openwisp/netjsonconfig #189: [compatiblity] Jinja 3.0 changes PackageLoader usage

atb00ker, 2021-05-14:
The [PackageLoader usage](https://github.com/openwisp/netjsonconfig/blob/master/netjsonconfig/backends/base/renderer.py#L22) would break in Jinga 3.0 (which is now used by openwisp-users on master branch); hence we should upgrade this.

nemesisdesign, 2021-05-14:
> The [PackageLoader usage](https://github.com/openwisp/netjsonconfig/blob/master/netjsonconfig/backends/base/renderer.py#L22) would break in Jinga 3.0 (which is now used by openwisp-users on master branch);

@atb00ker are you sure of this? I mean, openwisp-users AFAIK does not depend on Jinja, could you point out to me the line of code where it is being used?

atb00ker, 2021-05-15:
Oops, I meant openwisp-user installs Jinja 3.0, I have to manually downgrade it to avoid this error in openwisp-controller.
In the future, openwisp-controller will also install Jinja 3.0 then we will see the error shown in the screenshot!

PR #204 (2022-02-04): [deps] Upgrade to Jinja 3.0 #189
Closes #189
Tested with `openwisp-controller`, Jinja2~=3.0.0

[image]

[image]
commit ca737e32 (2022-02-04): [deps] Upgrade to Jinja 3.0 #189

Closes #189
closed 2022-02-04 by nemesisdesign