openwisp/netjsonconfig #195: Add support for DSA

nickberry17, 2021-08-23:
DSA is becoming the default for many OpenWrt targets in the upcoming stable release.  OpenWISP does not yet support DSA.

Docs
https://openwrt.org/docs/guide-user/network/dsa/start

nemesisdesign, 2022-03-04:
What the openwrt docs do not explain is that with the new syntax, we have to move any bridge related configuration to the "device" config types. However, since we need to be backward compatibile with the older versions of OpenWrt, I think we need to continue also doing what we are doing now (support both).

nemesisdesign, 2022-03-14:
For now we will focus on supporting the new bridge definition.

Defining DSA vlans can then be done with this custom config:

```
{
    "network": [
        {
            "config_name": "bridge-vlan",
            "config_value": "vlan1",
            "vlan": 1,
            "ports": [
                "lan1:u*",
                "lan2:u*",
                "lan3:u*",
                "lan4:u*"
            ]
        }
    ]
}
```

In the next release we can add a schema for this new syntax but before doing so we have to discuss a bit more and moreover wait for DSA to be implemented in all the devices, collect more feedback and make sure we implement it correctly.

nemesisdesign, 2022-03-14:
There's a dedicated issue for the new bridge syntax definition: https://github.com/openwisp/netjsonconfig/issues/195.

nemesisdesign, 2022-03-18:
I tried reading the examples and docs in the [OpenWrt wiki](https://openwrt.org/docs/guide-user/network/dsa/dsa-mini-tutorial) but it's not really clear to me, moreover the last example is not clear at all.

I cannot implement something if I don't understand the schema of the configuration, what the possible values are, the wiki mentions only examples but there's no explanation of the schema and possible values the configuration can have.

I hope someone can help clarify the situation a bit, but until then it will be pretty hard to implement something which is not even documented, nor I know at what source code to look at to figure it out.

nemesisdesign, 2022-03-18:
Maybe this video explains it better, I will look into it: https://www.youtube.com/watch?v=qeuZqRqH-ug.

nikolya-prodigy, 2022-06-07:
[image]

That's so much to draw to get

config bridge-vlan 'bridge_vlan10'
	option device 'br0'
	list ports 'wan:t'
	list ports 'lan1:t'
	list ports 'lan2:t'
	list ports 'lan3:t'
	list ports 'lan4:t'
	option vlan '10'

nemesisdesign, 2022-06-11:
Exampe custom syntax provided by @nikolya-prodigy to achieve what he wrote above:

```
{
    "interfaces": [
        {
            "type": "ethernet",
            "name": "br0.10",
            "mtu": 1500,
            "disabled": false,
            "network": "br0_vlan10",
            "mac": "",
            "autostart": true,
            "addresses": [
                {
                    "proto": "dhcp",
                    "family": "ipv4"
                }
            ]
        }
    ],
    "network": [
        {
            "config_name": "bridge-vlan",
            "config_value": "bridge_vlan10",
            "device": "br0",
            "vlan": 10,
            "ports": [
                "wan:t",
                "lan1:t",
                "lan2:t",
                "lan3:t",
                "lan4:t"
            ]
        }
    ]
}```

pandafy, 2023-03-31:
[OpenWrt documentation for VLAN configuration](https://openwrt.org/docs/guide-user/network/vlan/switch_configuration)

closed 2024-11-04 by nemesifier

Freak4915, 2025-07-27:
This appears to be added to the template level, but doesn't carry over to the final configuration.

For example, I create a template and include the VLAN Filtering, add VLAN 1 and the corresponding ports. Click Preview config and see the correct:

config bridge-vlan 'vlan_br_lan_1'
	option device 'br-lan'
	list ports 'lan1:t*'
	list ports 'lan2:t*'
	list ports 'lan3:t*'
	option vlan '1'

However, go to Devices and updated the device with the new template and click preview configuration, the above is missing and therefore never applies.

My only work around is to apply all the configurations and then copy them from the /etc/config/network file and then make a template for "files" to override the system file to include the above.

commit ab991d34 (2023-05-05): [feature] Added support for DSA #195

- Added VLAN 802.1q and VLAN 802.1ad interfaces

Closes #195
commit 593ac3b1 (2024-01-22): [feature] Added support for DSA #195

- Added VLAN 802.1q and VLAN 802.1ad interfaces

Closes #195
commit f5115c69 (2024-11-04): [feature] Added support for DSA #195

- Added VLAN 802.1q and VLAN 802.1ad interfaces

Closes #195