openwisp/netjsonconfig #208: [feature] Add ZeroTier backend

pandafy, 2022-02-16:
Add a ZeroTier backend that generated network configuration acceptable by REST API endpoints of ZeroTier controller.

Following fields should be supported by the backend. The schema is taken from OpenAPI specification of ZeroTier Controller REST API available at https://docs.zerotier.com/openapi/centralv1.json

```json 
"NetworkConfig": {
        "type": "object",
        "properties": {
          "id": {
            "type": "string",
            "example": "8056c2e21c000001",
            "description": "Network ID",
            "readOnly": true
          },
          "creationTime": {
            "type": "integer",
            "format": "int64",
            "example": 1442292672978,
            "description": "Time the network was created",
            "readOnly": true
          },
          "capabilities": {
            "type": "array",
            "description": "Array of network capabilities",
            "items": {
              "type": "object"
            }
          },
          "dns": {
            "type": "object",
            "$ref": "#/components/schemas/DNS"
          },
          "enableBroadcast": {
            "type": "boolean",
            "example": true,
            "description": "Enable broadcast packets on the network"
          },
          "ipAssignmentPools": {
            "type": "array",
            "description": "Range of IP addresses for the auto assign pool",
            "items": {
              "$ref": "#/components/schemas/IPRange"
            }
          },
          "lastModified": {
            "type": "integer",
            "format": "int64",
            "example": 1588184318235,
            "description": "Time the network was last modified",
            "readOnly": true
          },
          "mtu": {
            "type": "integer",
            "example": 2800,
            "description": "MTU to set on the client virtual network adapter"
          },
          "multicastLimit": {
            "type": "integer",
            "example": 32,
            "description": "Maximum number of recipients per multicast or broadcast. Warning - Setting this to 0 will disable IPv4 communication on your network!"
          },
          "name": {
            "type": "string",
            "example": "My ZeroTier Network"
          },
          "private": {
            "type": "boolean",
            "example": true,
            "description": "Whether or not the network is private.  If false, members will *NOT* need to be authorized to join."
          },
          "routes": {
            "type": "array",
            "items": {
              "$ref": "#/components/schemas/Route"
            }
          },
          "rules": {
            "type": "array",
            "items": {
              "type": "object"
            }
          },
          "tags": {
            "type": "array",
            "items": {
              "type": "object"
            }
          },
          "v4AssignMode": {
            "$ref": "#/components/schemas/IPV4AssignMode"
          },
          "v6AssignMode": {
            "$ref": "#/components/schemas/IPV6AssignMode"
          }
        }
      },
```

Aryamanz29, 2023-05-30:
OpenAPI specification for Self-hosted ZeroTier controllers (https://docs.zerotier.com/service/v1/):

```json
{
  "ControllerNetwork": {
    "type": "object",
    "properties": {
      "id": {
        "type": "string",
        "readOnly": true,
        "example": "3e245e31af000001"
      },
      "nwid": {
        "type": "string",
        "readOnly": true,
        "example": "3e245e31af000001"
      },
      "objtype": {
        "type": "string",
        "example": "network",
        "readOnly": true
      },
      "name": {
        "type": "string",
        "example": "my-cool-network"
      },
      "creationTime": {
        "type": "number",
        "example": 1623101592,
        "readOnly": true
      },
      "private": {
        "type": "boolean"
      },
      "enableBroadcast": {
        "type": "boolean"
      },
      "v4AssignMode": {
        "type": "object",
        "properties": {
          "zt": {
            "type": "boolean"
          }
        }
      },
      "v6AssignMode": {
        "type": "object",
        "properties": {
          "6plane": {
            "type": "boolean"
          },
          "rfc4193": {
            "type": "boolean"
          },
          "zt": {
            "type": "boolean"
          }
        }
      },
      "mtu": {
        "type": "integer",
        "example": 2800
      },
      "multicastLimit": {
        "type": "integer",
        "example": 32
      },
      "revision": {
        "type": "integer",
        "example": 1,
        "readOnly": true
      },
      "routes": {
        "type": "array",
        "items": {
          "type": "object",
          "properties": {
            "target": {
              "type": "string",
              "example": "192.168.192.0/24"
            },
            "via": {
              "type": "string",
              "example": "192.168.192.1",
              "nullable": true
            }
          }
        }
      },
      "ipAssignmentPools": {
        "type": "array",
        "items": {
          "type": "object",
          "properties": {
            "ipRangeStart": {
              "type": "string",
              "example": "192.168.192.1"
            },
            "ipRangeEnd": {
              "type": "string",
              "example": "192.168.192.254"
            }
          }
        }
      },
      "rules": {
        "type": "array",
        "items": {
          "type": "object"
        }
      },
      "capabilities": {
        "type": "array",
        "items": {
          "type": "object"
        }
      },
      "tags": {
        "type": "array",
        "items": {
          "type": "object"
        }
      },
      "remoteTraceTarget": {
        "type": "string",
        "example": "7f5d90eb87"
      },
      "remoteTraceLevel": {
        "type": "integer"
      }
    }
  }
}
```

Aryamanz29, 2023-06-01:
[image]

I have summarized what I found while researching ZeroTier self-hosting and network configuration. Here's an overview:

When it comes to self-hosting, we have two types of ZeroTier infrastructure:

**Network Controllers** (Focused on currently)

- Controllers handle [Virtual Layer 2](https://docs.zerotier.com/zerotier/manual#22vl2theethernetvirtualizationlayeraname2_2a). Controllers do not handle traffic.

- ZeroTier Inc. runs controllers with a web UI and API at [ZeroTier Central](https://my.zerotier.com/).

We can use [self-hosted controllers](https://docs.zerotier.com/self-hosting/network-controllers), but ZeroTier currently does not offer a web UI for self-hosting. Therefore, we can use an [open-source controller web UI provided by Key Networks](https://github.com/key-networks/ztncui) for network configuration. Additionally, we can utilize the [service API](https://docs.zerotier.com/service/v1/) or directly generate JSON configuration (as discussed below).

**Roots a.k.a "Moons"**

- Roots handle [Virtual Layer 1](https://docs.zerotier.com/zerotier/manual#211networktopologyandpeerdiscoveryaname2_1_1a). ZeroTier Inc. hosts the roots.

- Peer-to-peer connections are set up via the roots. If peer-to-peer connections cannot be established, traffic is "relayed" through these roots.

	**When are they required?**

	- They are required when we want to avoid relying entirely on ZeroTier Inc.'s data center and need to connect devices without internet access (often used within organizations).

	**Drawbacks**

	- Only manual configuration is available; ZeroTier does not provide API support for management.

#### Network Controllers Configuration:

Initially, when I was implementing the ZeroTier backend, I thought it stored network configurations in the `network.d` directory. However, I later discovered that this directory is used for managing network configurations that the controller/node **can join or has already joined** (in `<network_id>.conf` **binary format**). On the other hand, the `controller.d/network/` directory contains network configurations that were created by the controller (in `<network_id>.json` **JSON format**). This is what the ZeroTier backend should render in order to manage controller networks.

##### Possible Solution

1. We can add ZeroTier `authtoken_secret` to the ZeroTier backend schema and backend will render the configuration as a  `cURL` request shown below:

	```bash
	# Create a network
	curl -X POST \
  	--header 'Accept: application/json' \
  	--header 'Content-Type: application/json' \
  	--header 'X-ZT1-Auth: <authtoken_secret>' \
  	--data '<network_configuration.json>' \
  	'http://localhost:9993/controller/network/<node_id>______'
	```
	
	**Note:** We can retrieve controller `<node_id>` beforehand by calling `GET http://localhost:9993/status`.

2. Another approach is to only render the `network_configuration.json`, and users can use that `JSON` file to create new networks by saving it under the `controller.d/network/` directory, followed by restarting the Zerotier-one service (i.e. `systemctl restart zerotier-one`) 

	**Note:** This method is typically used when we don't want to create a network with the service API.

##### Sample `network_configuration.json`: https://pastebin.com/raw/ugzcynt5

#### So now my question is, which solution should we choose to proceed further with the ZeroTier VPN backend?

##### References:

https://docs.zerotier.com/self-hosting/introduction
https://docs.zerotier.com/zerotier/moons/
https://docs.zerotier.com/zerotier/troubleshooting
https://discuss.zerotier.com/t/diffenrent-local-conf-configuration-for-each-network/13343
https://discuss.zerotier.com/t/flow-rules-for-self-hosted-controller/514/2

Aryamanz29, 2023-06-01:
>I have one question: Is it possible to change all configuration options (present in network configuration in controller.d/network/) from REST API itself? If not, what happens when we add that parameter in the HTTP request? Will the controller raise an error? I am asking this because we will be using the configuration generated here for making HTTP requests in openwisp-controller.

@pandafy, the majority of options in the configuration can be modified using the REST API. If we include a parameter that cannot be updated or if its type does not conform to the OpenAPI schema, the controller will simply ignore that parameter. In such cases, a status 200 will be returned, indicating success, and only the parameters that adhere to the OpenAPI schema of the ZeroTier service API will have their values updated.

Reference: https://github.com/zerotier/ZeroTierOne/blob/dev/service/README.md#network-virtualization-service-api

nemesisdesign, 2023-06-02:
> We should just generate the network configuration in the JSON format and render that. What do you think @nemesisdesign?

I think this works. We have to find a solution which is the best compromise between making it work and making it sane for the user to work with.

commit 116c0536 (2023-05-30): [feature] Added ZeroTier backend #208

Closes #208
commit 125b8973 (2023-06-11): [feature] Added ZeroTier backend #208

Closes #208
commit 3ec95168 (2023-06-16): [feature] Added ZeroTier backend #208

Closes #208
PR #272 (2023-06-16): [feature] Added ZeroTier backend #208
#### Prerequisites:

1. Install [Zerotier](https://www.zerotier.com/download/)

2. Install [Ztncui - Web Interface](https://key-networks.com/ztncui/)

3. Get your Zerotier Service API token `sudo cat /var/lib/zerotier-one/authtoken.secret`

#### Tested with openwisp-controller

- **Prototype Branch**:  https://github.com/openwisp/openwisp-controller/tree/zerotier-prototype

https://github.com/openwisp/netjsonconfig/assets/56113566/5787c6a8-1cd0-4d00-8d7c-a118252bed60

**Todos**

- [x] Improve and extend the existing ZeroTier backend schema using the OpenAP [... cut]
closed 2023-10-11 by nemesifier