openwisp/netjsonconfig #227: [bug] The LED mode is supposed to be a list not an option

mfoxworthy, 2022-05-03:


nemesisdesign, 2022-05-03:
@mfoxworthy see this passage from https://openwrt.org/docs/guide-user/base-system/led_configuration 

> mode | string | yes | (none) | One or more of link, tx, or rx, seperated by spaces
> -- | -- | -- | -- | --

Mode: One or more of link, tx, or rx, seperated by spaces.

> 
> ### WLAN led
> 
> ```
> config 'led' 'wlan_led'
> 	option 'name'           'WLAN'
> 	option 'sysfs'          'tl-wr1043nd:green:wlan'
> 	option 'trigger'        'netdev'
> 	option 'dev'            'wlan0'
> 	option 'mode'           'link tx rx'
> ```

Therefore, in your configuration, set mode as `"link tx rx"` and according to the OpenWrt docs it should work.

@okraits 

> The relevant project for this issue is https://github.com/openwisp/netjsonconfig. Could you please open an issue in that project?

We can move the issue without asking users to close and recreate it.

okraits, 2022-05-03:
> We can move the issue without asking users to close and recreate it.

I didn't know that - thank you for the hint :-)

mfoxworthy, 2022-05-03:
> @mfoxworthy see this passage from https://openwrt.org/docs/guide-user/base-system/led_configuration
>
>
> Mode: One or more of link, tx, or rx, seperated by spaces.
>
>
> Therefore, in your configuration, [...]

You are correct. I just tested this with success. Sorry for the bother. Maybe I should read the docs some more. 😳

closed 2022-05-03 by mfoxworthy