openwisp/netjsonconfig #228: [bug] Adding modem manager network interface doesn't work with OpenWRT 21.02.2/3 with both PPP and QMI drivers loaded.

mfoxworthy, 2022-05-03:


nemesisdesign, 2022-05-05:
@mfoxworthy can you please provide the full `/etc/config/network` file and remove any sensitive info?

mfoxworthy, 2022-05-05:
```
config interface 'wifinet0'
	option auto '1'
	option enabled '1'
	option ifname 'wlan0'
	option mtu '1500'
	option proto 'none'

config interface 'wifinet1'
	option auto '1'
	option enabled '1'
	option ifname 'wlan1'
	option mtu '1500'
	option proto 'none'

config interface 'loopback'
	option device 'lo'
	option proto 'static'
	option ipaddr '127.0.0.1'
	option netmask '255.0.0.0'

config globals 'globals'
	option ula_prefix 'fd44:3cbe:ebc3::/48'

config device 'device_lan'
	option name 'br-lan'
	option type 'bridge'
	option hash_max '512'
	option igmp_snooping '1'
	option last_member_interval '100'
	option mtu '1500'
	option multicast_querier '1'
	option query_interval '12500'
	option query_response_interval '1000'
	option robustness '2'
	list ports 'eth0'

config interface 'lan'
	option device 'br-lan'
	option proto 'static'
	option ipaddr 'XXXX'
	option netmask '255.255.255.0'
	option ip6assign '60'
	option auto '1'
	option enabled '1'

config interface 'wan'
	option device 'eth1'
	option proto 'dhcp'
	option metric '10'

config interface 'wan6'
	option device 'eth1'
	option proto 'dhcpv6'

config switch 'switch0'
	option enable_vlan '1'
	option name 'switch0'
	option reset '1'

config switch_vlan 'switch0_vlan1'
	option device 'switch0'
	option ports '3 4 0'
	option vlan '1'

config interface 'lte'
	option apn 'XXXX'
	option device '/sys/devices/platform/soc/8af8800.usb3/8a00000.dwc3/xhci-hcd.0.auto/usb1/1-1'
	option enabled '1'
	option iptype 'ipv4'
	option proto 'modemmanager'
	option signalrate '5'
```
Working:

```
      usb-modeswitch
      kmod-mii
      kmod-usb-net
      kmod-usb-wdm
      kmod-usb-net-qmi-wwan
      uqmi
```
Breaks it:
```
      kmod-usb-serial-option
      kmod-usb-serial
      kmod-usb-serial-wwan
```
The serial modules allow us to use stty and serial commands to check for messages and balance. We can see messages with the uqmi application but we integrated stty with our scripts. We can't get usage data with uqmi.

nemesisdesign, 2022-05-05:
@mfoxworthy, so, if you try the following:

```
config interface 'lte'
	option apn 'XXXX'
	option device '/sys/devices/platform/soc/8af8800.usb3/8a00000.dwc3/xhci-hcd.0.auto/usb1/1-1'
	option enabled '1'
	option iptype 'ipv4'
	option proto 'modemmanager'
	option signalrate '5'
        option mtu '1420'
```

Is the MTU value honored?

mfoxworthy, 2022-05-05:
No. I just tried it. Here is the link to the parameters modem manager will accept:

https://openwrt.org/docs/guide-user/network/wan/wwan/modemmanager

When using qmi mode the kmod driver creates an interface called wwan0. If we want to set the MTU o that device, we'd need a option name 'wwan0' section with option mtu '1428':

```
config device
	option name 'wwan0'
	option mtu '1428'
	option ipv6 '0'
```
Openwrt only uses device sections now. With qmi /sys/devices/platform/soc/20980000.usb/usb1/1-1/1-1.2/1-1.2.1 translates to /dev/cdc-wdm0 which is tied to wwan0.

With the serial drivers a device with the name ppp0 is created and a completely different syspath is used in modem manager.

nemesisdesign, 2022-05-05:
So, to summarize for @pandafy, I think the following NetJSON:

```json
{
    "interfaces": [
        {
            "type": "modem-manager",
            "apn": "brodband",
            "device": "/sys/devices/platform/soc/8af8800.usb3/8a00000.dwc3/xhci-hcd.0.auto/usb1/1-1",
            "iptype": "ipv4",
            "name": "wwan0",
            "disabled": false,
            "signalrate": 5,
            "mtu": 1428
        }
    ]
}
```

Shall generate the following UCI config on OpenWrt:

```
config interface 'wwan0'
	option apn 'brodband'
	option device '/sys/devices/platform/soc/8af8800.usb3/8a00000.dwc3/xhci-hcd.0.auto/usb1/1-1'
	option enabled '1'
	option iptype 'ipv4'
	option proto 'modemmanager'
	option signalrate '5'

config device `device_wwan0`
	option name 'wwan0'
	option mtu '1428'
```

@mfoxworthy do you confirm? We'll implement a fix and ask you to test it.

mfoxworthy, 2022-05-05:
> So, to summarize for @pandafy, I think the following NetJSON:
>
> ```json
> {
>     "interfaces": [
>         {
>             "type": "modem-manager",
>             "apn": "brodband",
>             "device": [...]

I confirm that will work. Most excellent!

pandafy, 2022-05-06:
Hey @mfoxworthy! I have created a fix for this issue in #229. Can you help us test this? 

If you have installed OpenWISP using ansible-openwisp2, then follow these instructions to install netjsonconfig from that branch: 

1. SSH into the remote machine where OpenWISP is installed 
2. Elevate to superuser status using `sudo su`
3. Activate the python virtual environment 
```
source /opt/openwisp2/env/bin/activate
```
4. Install netjsonconfig from the `issues/228-modem-manager-dsa` branch 
```
pip install -U --no-cache-dir https://github.com/openwisp/netjsonconfig/tarball/issues/228-modem-manager-dsa
```
5. Restart all services to use the updated netjsonconfig module
```
supervisorctl restart all
```

After following above steps, you can log into OpenWISP and create configuration for your devices. I have also added the `signalrate` option in the schema.

mfoxworthy, 2022-05-10:
> supervisorctl restart all

```
config interface 'lte'
        option device '/sys/devices/platform/soc/8af8800.usb3/8a00000.dwc3/xhci-hcd.0.auto/usb1/1-1'
        option enabled '1'
        option iptype 'ipv4'
        option proto 'modemmanager'
        option signalrate '5'
        option apn 'broadband'
        option lowpower '0'
        option metric '20'

config device 'device_lte'
        option mtu '1500'
        option name 'wwan0'
```
New branch was installed and new configuration was added.

**Confirmed to work! Great Job and fast turn around! You guys rock!**

BTW, I don't know if you have looked into it but option ifname has gone away and option device replaced it. Luci will fix it and it isn't really broken but just thought I'd mention it. I wouldn't have even noticed if I didn't have some users who use Luci. This screen will pop up:

[image]

```config interface 'wifinet0'
	option auto '1'
	option enabled '1'
        option ifname 'wlan0' <--- should be "option device 'wlan0'" for OpenWrt 21.02+
	option mtu '1500'
	option proto 'none'

nemesisdesign, 2022-05-10:
> ```
> config interface 'lte'
>         option device '/sys/devices/platform/soc/8af8800.usb3/8a00000.dwc3/xhci-hcd.0.auto/usb1/1-1'
>         option enabled '1'
>         option iptype 'ipv4'
>         option [...]

Thanks for confirming!

> BTW, I don't know if you have looked into it but option ifname has gone away and option device replaced it. Luci will fix it and it isn't really broken but just thought I'd mention it. I wouldn't have even noticed if I didn't have some users who use Luci. This screen will pop up:
> 
> [image]
> 
> ```
> 	option auto '1'
> 	option enabled '1'
>         option ifname 'wlan0' <--- should be "option device 'wlan0'" for OpenWrt 21.02+
> 	option mtu '1500'
> 	option proto 'none'
> ```

@mfoxworthy can you please open a new issue for this in which you specify the configuration you are using for that?

mfoxworthy, 2022-05-10:
I will. I was thinking you may have already seen this so I waited for your confirmation.

PR #229 (2022-05-10): [fix] Fixed modem-manager configuration for OpenWrt 21 #228
Closes #228
commit 858aa194 (2022-05-10): [fix] Fixed modem-manager configuration for OpenWrt 21 #228

- Fixed modem-manager configuration for OpenWrt 21
- Added missing signalrate option for modemmanager interface

Closes #228
closed 2022-05-10 by nemesisdesign