openwisp/netjsonconfig #230: [bug] ifname option should now be using the device option for OpenWrt 21.02+

mfoxworthy, 2022-05-10:
Option ifname has gone away and option device replaced it. Luci will fix it and it isn't really broken but just thought I'd mention it. I wouldn't have even noticed if I didn't have some users who use Luci. This screen will pop up:

[image]

```config interface 'wifinet0'
	option auto '1'
	option enabled '1'
        option ifname 'wlan0' <--- should be "option device 'wlan0'" for OpenWrt 21.02+
	option mtu '1500'
	option proto 'none'
```
This particular instance was seen when using the "Wireless Interface" interface template.

I believe the "name" key is getting translated as "ifname" and it should be translated to "device" in OpenWRT 21.02.X as per the OpenWRT documentation found in this link:

[[https://openwrt.org/docs/guide-user/base-system/basic-networking](https://github.com/openwisp/netjsonconfig/issues/url)](url)

```
    "interfaces": [
        {
            "wireless": {
                "network": [
                    "lan"
                ],
                "mode": "access_point",
                "radio": "radio0",
                "ack_distance": 0,
                "rts_threshold": 0,
                "frag_threshold": 0,
                "ssid": "{{ def_ssid }}",
                "hidden": false,
                "wds": false,
                "encryption": {
                    "protocol": "wpa2_personal_mixed",
                    "key": "SturdyNet1!",
                    "disabled": false,
                    "cipher": "ccmp",
                    "ieee80211w": "1"
                },
                "wmm": true,
                "isolate": false,
                "ieee80211r": false,
                "reassociation_deadline": 1000,
                "ft_psk_generate_local": false,
                "ft_over_ds": true,
                "rsn_preauth": false,
                "macfilter": "disable",
                "maclist": []
            },
            "type": "wireless",
            "name": "wlan0",
            "mtu": 1500,
            "disabled": false,
            "network": "wifinet0",
            "mac": "",
            "autostart": true,
            "addresses": []
        },
```

nemesisdesign, 2022-05-10:
@mfoxworthy it would be great if you could test the fix by installing the tarball from `https://github.com/openwisp/netjsonconfig/tarball/issues/230-wifi-dsa`, try again and let me know if you still get warnings.

mfoxworthy, 2022-05-11:
This fix was installed and confirmed to work. Thanks! You guys are fast!

```
config device 'device_wifinet0'
	option mtu '1500'
	option name 'wlan0'

config interface 'wifinet0'
	option auto '1'
	option device 'wlan0'
	option enabled '1'
	option proto 'none'

config device 'device_wifinet1'
	option mtu '1500'
	option name 'wlan1'

config interface 'wifinet1'
	option auto '1'
	option device 'wlan1'
	option enabled '1'
	option proto 'none'
```

nemesisdesign, 2022-05-11:
@mfoxworthy thanks for confirming! I hope more users like you will help us to find ways to improve the compatibility of OpenWISP with the new DSA syntax of OpenWrt, with a bit more collaboration from everyone we can do very well :+1: :clap:!

mfoxworthy, 2022-05-11:
We are very excited about what OpenWISP is doing. You should see a lot more cases from our testing and development. Maybe we can talk offline about what we are doing with your system. It's a fairly unique use case. Great job and I really look forward to the new enhancements! 👊

commit af9e8b5d (2022-05-11): [fix] Added wireless interface to DSA types #230

Fixes #230
closed 2022-05-11 by nemesisdesign