openwisp/netjsonconfig #238: WiFi Station BSSID does not accept variable properly

lpalgarvio, 2022-06-18:
Hello,
Everytime i try to add a variable to BSSID field on a WiFi interface configured as station,
It breaks the JSON configuration after/during saving.

ie,
```
        {
            "wireless": {
                "network": [
                    "lan"
                ],
                "mode": "station",
                "radio": "radio1",
                "ssid": "{{wireless_wlan0sta_ssid}}",
                "bssid": "00:11:22:33:44:55",
                "wds": true,
                "encryption": {
                    "protocol": "wpa2_personal",
                    "key": "{{wireless_wlan0sta_key}}",
                    "cipher": "ccmp",
                    "ieee80211w": "1"
                }
            },
            "type": "wireless",
            "name": "radio0sta",
            "network": "wlan0sta"
        },
```

becomes

```
        {
            "type": "ethernet",
            "name": "radio1sta",
            "network": "wlan1sta",
            "wireless": {
                "network": [
                    "lan"
                ],
                "mode": "station",
                "radio": "radio1",
                "ssid": "{{wireless_wlan1sta_ssid}}",
                "bssid": "{{wireless_wlan1sta_bsid}}",
                "wds": true,
                "encryption": {
                    "protocol": "wpa2_personal",
                    "key": "{{wireless_wlan1sta_key}}",
                    "cipher": "ccmp",
                    "ieee80211w": "1"
                }
            }
        }
```

I haven't looked at the code but i can try make a PR sometime later if i find the issue

nemesisdesign, 2022-06-19:
@lpalgarvio I am not sure I understood this, I shall try asap.

lpalgarvio, 2022-06-20:
The cause is likely this:

```
        "bssid_wireless_property": {
            "properties": {
                "bssid": {
                    "type": "string",
                    "title": "BSSID",
                    "pattern": MAC_PATTERN_BLANK,
                    "maxLength": 17,
                    "propertyOrder": 4,
                },
            }
        },
```

the pattern and maxLength being used there will probably not allow for a variable.
Not sure that's possible without refactoring, but should allow either macaddress or variable with macaddress.

Other places where this can be a problem:
```
                "mac": {
                    "type": "string",
                    "title": "MAC address",
                    "description": "if specified overrides default macaddress for this interface",
                    "pattern": MAC_PATTERN_BLANK,  # can be empty
                    "maxLength": 17,
                    "propertyOrder": 3,
                },

        "ssid_wireless_property": {
            "required": ["ssid"],
            "properties": {
                "ssid": {
                    "type": "string",
                    "title": "SSID",
                    "maxLength": 32,
                    "propertyOrder": 3,
                } 
            },
        },

        "adhoc_wireless_settings": {
            "title": "Adhoc",
            "allOf": [
                {
                    "required": ["bssid"],
                    "properties": {
                        "mode": {"enum": ["adhoc"]},
                        "bssid": {"pattern": MAC_PATTERN, "minLength": 17},
                    },
                },
                {"$ref": "#/definitions/base_wireless_settings"},
                {"$ref": "#/definitions/ssid_wireless_property"},
                {"$ref": "#/definitions/bssid_wireless_property"},
                {"$ref": "#/definitions/encryption_wireless_property_mesh"},
            ],
        },
```

Maybe we can just fix the formula here:
```
MAC_PATTERN = '([0-9A-Fa-f]{2}[:-]){5}([0-9A-Fa-f]{2})'
MAC_PATTERN_BLANK = '^({0}|)$'.format(MAC_PATTERN)
```

and remove maxLength.

lpalgarvio, 2022-06-20:
The new regex expression suggests this will work fine:

https://regex101.com/
`([0-9A-Fa-f]{2}[:-]){5}([0-9A-Fa-f]{2})|([{]{2}.*[}]{2})`

```
EE:08:6A:A6:8D:D0
EE-08-6A-A6-8D-D0
{{wireless_wlan0sta_bssid}}
```

lpalgarvio, 2022-06-20:
> @lpalgarvio I am not sure I understood this, I shall try asap.

PR provided :)

lpalgarvio, 2022-06-20:
intended result
[image]