openwisp/netjsonconfig #242: Failing to create a template with more than one bridged interface

okraits, 2022-10-24:
I'm trying to create a template with 3 bridged interfaces. We are using the old pre-DSA syntax.

If I save the template, the second and third bridged interface are reverted to network interfaces. Here is the raw output of the template after saving:
[image]

Nevertheless, the template seems to be saved with the bridged interfaces in the database, because the device config contains the three bridged interfaces:
[image]

nemesisdesign, 2022-11-01:
@okraits can you please share the NetJSON configuration that can be used to replicate the issue and explain the steps, eg:

- click on add a new template
- paste this NetJSON config
- hit save

Expected result: ...
Actual result: ...

nemesisdesign, 2023-06-14:
Closing for now, if you have more info on how to replicate this bug please reopen.

closed 2023-06-14 by nemesisdesign