openwisp/netjsonconfig #250: OpenVPN tls_auth enabling behavior

kosli, 2023-01-22:
What is the expected behavior if the tls-auth (#174) on the OpenWISP server config gets activated **after** the client config template has already been generated? Should the client template automatically be updated or how can the tls-auth config be manually added to the client template?

nemesisdesign, 2023-01-23:
Not sure I fully understand the question, I think you mean how to enable tls-auth on already existing templates.
You can do so via advanced mode or from the upper right button in the configuration editor which lists the available options which can be enabled.
Please ask further questions in the https://gitter.im/openwisp/general or on the mailing list, we reserve github issues for development work only (bugs, features, changes, maintenance).

closed 2023-01-23 by nemesisdesign

kosli, 2023-01-23:
Thanks @nemesisdesign  I have had manually updated the template. So there is no way to automatically update the template if the "VPN Servers" have been updated..