openwisp/netjsonconfig #252: [feature] add support for tls-crypt instead of tls-auth

kosli, 2023-02-16:
OpenVPN 2.5 onwards supports the tls-crypt option instead of tls-auth.
It would be great if OpenWISP would support this feature instead of just tls-auth.

From the [OpenVPN 2.5. manual](https://openvpn.net/community-resources/reference-manual-for-openvpn-2-5/):
- Encrypt and authenticate all control channel packets with the key from keyfile. (See --tls-auth for more background.)
- Encrypting (and authenticating) control channel packets:
  - provides more privacy by hiding the certificate used for the TLS connection,
  - makes it harder to identify OpenVPN traffic as such,
  - provides "poor-man's" post-quantum security, against attackers who will never know the pre-shared key (i.e. no forward secrecy).
- In contrast to --tls-auth, --tls-crypt does not require the user to set --key-direction.

kosli, 2023-02-16:
OpenVPN 2.5 would even support client-specific tls-crypt keys, see
**Client-specific tls-crypt keys (--tls-crypt-v2)**
`tls-crypt-v2` adds the ability to supply each client with a unique tls-crypt key. This allows large organisations and VPN providers to profit from the same DoS and TLS stack protection that small deployments can already achieve using tls-auth or tls-crypt.

amanbn8, 2026-10-04:
Hi, I'd like to take this one.

Plan, mirroring how `tls_auth` is handled today:

- schema: new optional `tls_crypt` string field (textarea) in the OpenVPN backend, next to `tls_auth`;
- converter: when the field holds the key material rather than a path, write it to `<ca dir>/<dev>_tls_crypt.key` the same way `__add_tls_auth_key` does and render `tls-crypt <path>` (no key direction, tls-crypt keys are symmetric);
- `auto_client()`: copy the key material into the generated client config, so clients created by openwisp-controller pick it up automatically;
- validation: reject configurations that set both `tls_auth` and `tls_crypt`, since OpenVPN refuses to start with both (`--tls-auth and --tls-crypt are mutually exclusive`);
- the OpenWrt backend needs no special handling (`tls_crypt` is already in the openvpn package's option list) and the parser picks the directive up through the generic key/value path; docs and tests included.

I'd leave `tls-crypt-v2` out of this PR: it needs per-client keys, so `auto_client()` would need a new argument to receive them. Better handled separately once the basic option is in.

Unless you'd prefer a different approach, I'll open a PR referencing this issue.