openwisp/netjsonconfig #253: [BUG] Cannot have more than one wireguard peer

SerialVelocity, 2023-02-26:
Hey openwisp team,

It seems that it is not possible to have more than one wireguard peer when using the openwrt backend. I think the issue is that the peer is being named after the interface so each peer just overwrites the previous one:
https://github.com/openwisp/netjsonconfig/blob/1.0.2/netjsonconfig/backends/openwrt/converters/wireguard_peers.py#L20

nemesisdesign, 2023-02-28:
@SerialVelocity could you provide a NetJSON config sample which triggers this issue please?

SerialVelocity, 2023-03-05:
I haven't tested this, but constructed an example based on your docs (with an extra peer):
```
{
    "interfaces": [
        {
            "name": "wg",
            "type": "wireguard",
            "private_key": "QFdbnuYr7rrF4eONCAs7FhZwP7BXX/jD/jq2LXCpaXI=",
            "port": 51820,
            "mtu": 1420,
            "nohostroute": False,
            "fwmark": "",
            "ip6prefix": [],
            "addresses": [
                {
                    "proto": "static",
                    "family": "ipv4",
                    "address": "10.0.0.5/32",
                    "mask": 32,
                }
            ],
            "network": "",
        }
    ],
    "wireguard_peers": [
        {
            "interface": "wg",
            "public_key": "94a+MnZSdzHCzOy5y2K+0+Xe7lQzaa4v7lEiBZ7elVE=",
            "allowed_ips": ["10.0.0.1/32"],
            "endpoint_host": "wireguard.test.com",
            "endpoint_port": 51820,
            "preshared_key": "",
            "persistent_keepalive": 60,
            "route_allowed_ips": True,
        },
        {
            "interface": "wg",
            "public_key": "e8yh3kIrMjLVZPi5DS1BU3j3mWkYWrCXvnLSvNkW7HM=",
            "allowed_ips": ["10.0.0.2/32"],
            "endpoint_host": "wireguard2.test.com",
            "endpoint_port": 51820,
            "preshared_key": "",
            "persistent_keepalive": 60,
            "route_allowed_ips": True,
        }
    ]
}
```

closed 2023-05-04 by nemesisdesign

nemesisdesign, 2023-05-04:
@SerialVelocity fixed in the latest master.

commit c4cdf682 (2023-05-04): [fix] Fixed multiple wireguard peers bug #253

Closes #253
PR #268 (2023-05-04): [fix] Fixed multiple wireguard peers bug #253
Closes #253