openwisp/netjsonconfig #254: [BUG] Cannot use wpa3-personal for 802.11s mesh

SerialVelocity, 2023-02-26:
It seems wpa3-personal (not sure if mixed is possible as well) is missing from the schema:
https://github.com/openwisp/netjsonconfig/blob/1.0.2/netjsonconfig/schema.py#L400-L404

Is there a way of setting this manually and bypassing the checks? Usually, you can override fields by specifying "unknown properties" but the "encryption" key is re-used here.

nemesisdesign, 2023-02-28:
@SerialVelocity we need to add `{"$ref": "#/definitions/encryption_wpa3_personal"},` plus a couple of tests for rendering and parsing.

SerialVelocity, 2023-03-06:
I think another issue is the `2` has been removed from:
https://github.com/openwisp/netjsonconfig/blob/1.0.2/netjsonconfig/schema.py#L496

which affects wpa2/wpa3 mixed mode. Quite a big issue if you have multiple SSIDs because if you have a WPA3 SSID with required 802.11w, and a WPA2/WPA3 mixed mode SSID with optional 802.11w, then your speed will drop from ~770Mbps to ~150Mbps.

nemesisdesign, 2023-04-19:
@SerialVelocity can you look whether this patch correctly adds support for WPA3 in mesh mode?

https://github.com/openwisp/netjsonconfig/pull/262

> I think another issue is the `2` has been removed from: https://github.com/openwisp/netjsonconfig/blob/1.0.2/netjsonconfig/schema.py#L496
>
> which affects wpa2/wpa3 mixed mode. Quite a big issue if [...]

Doesn't mfp have to be optional in mixed mode? Can it be required and work on WPA2 too? Not sure I understood 100%.

SerialVelocity, 2023-04-25:
I think it adds it correctly. I'm running the docker images of openwisp so I'm not sure how to try deploying your patch.

Afaik, the "pmf is required" extension came before wpa3 was a thing:
 https://en.wikipedia.org/wiki/IEEE_802.11w-2009

I can't find anywhere that says pmf has to be optional for mixed mode as it can already be required for just wpa2. Could you point me to where you have seen that be the case? (I haven't explicitly tried connecting with wpa2 when mixed mode is enabled and pmfs are required)

nemesisdesign, 2023-04-29:
> I think it adds it correctly. I'm running the docker images of openwisp so I'm not sure how to try deploying your patch.
>
> Afaik, the "pmf is required" extension came before wpa3 was a thing: [...]

I can't find the information anymore right now, can you please create a separate issue for this other subject? We should look at what LuCI (the web interface of OpenWrt) does and just follow along.

SerialVelocity, 2023-04-30:
Done!

commit e392d175 (2023-04-19): [feature] Added wpa3-personal encryption for 802.11s mesh #254

Closes #254
PR #262 (2023-05-02): [feature] Added wpa3-personal encryption for 802.11s mesh #254
Closes #254
commit 3d105cda (2023-05-02): [feature] Added wpa3-personal for 802.11s mesh #254

Closes #254
closed 2023-05-02 by nemesisdesign