openwisp/netjsonconfig #289: [bug?] When adding a Wireless config, should it add an interface?

thiagohflima, 2023-11-30:
## Scenario

Consider the following network configuration:

- Two VLANs: 10 and 20
- Bridges named vlan10 using eth0.10 and vlan20 using eth0.20

The goal is to create two SSIDs for each VLAN:

- Wifi_VLAN10
- Wifi_VLAN20

## NetJSON Configuration

```json
[
  {
    "wireless": {
      "network": ["vlan10"],
      "mode": "access_point",
      "radio": "radio0",
      "ssid": "WifiVLAN10"
    },
    "type": "wireless",
    "name": "wlan0"
  },
  {
    "wireless": {
      "network": ["vlan20"],
      "mode": "access_point",
      "radio": "radio0",
      "ssid": "WifiVLAN20"
    },
    "type": "wireless",
    "name": "wlan1"
  }
]
```

This configuration generates the following (showing only wireless-related settings):

## Package Network

```
config device 'device_wlan0'
	option name 'wlan0'

config interface 'wlan0'
	option device 'wlan0'
	option proto 'none'

config device 'device_wlan1'
	option name 'wlan1'

config interface 'wlan1'
	option device 'wlan1'
	option proto 'none'
```

## Package Wireless

```
config wifi-iface 'wifi_wlan0'
	option device 'radio0'
	option ifname 'wlan0'
	option mode 'ap'
	option network 'vlan10'
	option ssid 'WifiVLAN10'

config wifi-iface 'wifi_wlan1'
	option device 'radio0'
	option ifname 'wlan1'
	option mode 'ap'
	option network 'vlan20'
	option ssid 'WifiVLAN20'
```

---

When manually configuring in OpenWRT, it is simplified as follows:

## Package Network
There's no config added in `network`

## Package Wireless
```
config wifi-iface 'wifi_wlan0'
	option device 'radio0'
	option mode 'ap'
	option network 'vlan10'
	option ssid 'WifiVLAN10'

config wifi-iface 'wifi_wlan1'
	option device 'radio0'
	option mode 'ap'
	option network 'vlan20'
	option ssid 'WifiVLAN20'
```

---

As observed, the manual configuration lacks the ifname attribute in the wireless package, and no interfaces are created in the network package. While the current configuration works, there's a consideration about simplifying the OpenWisp-generated config. Does this simplification make sense?

nemesifier, 2024-05-17:
I ran into a situation in which the additional configuration in `/etc/config/network` caused issues.
I am investigating the possibility of removing that.

nemesifier, 2024-06-07:
I don't think we can remove `ifname` from the wireless configuration for different reasons, it doens't hurt having it defined explicitly.
So I am focusing on removing the unnecessary definitions in the `/etc/config/network`.

nemesifier, 2024-07-03:
https://github.com/openwisp/netjsonconfig/pull/298 should fix this. I am testing it.

commit 3a6aa96d (2024-06-07): [change] Wireless: avoid unnecessary network config #289

Avoid generating unnecessary config in /etc/config/network
when defining wireless interfaces.

Closes #289
commit 3cb647b8 (2024-07-09): [change] Wireless: avoid unnecessary network config #289

Avoid generating unnecessary config in /etc/config/network
when defining wireless interfaces.

Closes #289
PR #298 (2024-07-09): [change] Wireless: avoid unnecessary network config #289
Avoid generating unnecessary config in /etc/config/network when defining wireless interfaces.

Closes #289
commit d0e9815f (2024-07-09): [change] Wireless: avoid unnecessary network config #289

Avoid generating unnecessary config in /etc/config/network
when defining wireless interfaces.

Closes #289
closed 2024-07-09 by nemesifier