openwisp/netjsonconfig #316: [bug] Modem-manager generates unwanted network device

nemesifier, 2024-09-03:
Continuation of #314, #298.

The following modem-manager conf:

```python
{
    "interfaces": [
        {
            "type": "modem-manager",
            "apn": "apn.operator.com",
            "pin": "1234",
            "device": "/sys/devices/platform/ahb/1b000000.usb/usb1/1-1",
            "username": "user123",
            "password": "pwd123456",
            "metric": 50,
            "lowpower": False,
            "name": "modem0",
            "mtu": 1500,
            "signalrate": 5,
        }
    ]
}
```

Generates:

```
    package network

    config device 'device_modem0'
            option mtu '1500'
            option name 'modem0'

    config interface 'modem0'
            option apn 'apn.operator.com'
            option device '/sys/devices/platform/ahb/1b000000.usb/usb1/1-1'
            option lowpower '0'
            option metric '50'
            option password 'pwd123456'
            option pincode '1234'
            option proto 'modemmanager'
            option signalrate '5'
            option username 'user123'
```

But I believe the following would be sufficient:

```
    package network

    config interface 'modem0'
            option apn 'apn.operator.com'
            option device '/sys/devices/platform/ahb/1b000000.usb/usb1/1-1'
            option lowpower '0'
            option metric '50'
            option password 'pwd123456'
            option pincode '1234'
            option proto 'modemmanager'
            option signalrate '5'
            option username 'user123'
```

pandafy, 2024-09-20:
@nemesifier if the expected configuration `ifname` and `mtu` fields are absent. Is that intentional?

nemesifier, 2024-09-20:
> @nemesifier if the expected configuration `ifname` and `mtu` fields are absent. Is that intentional?

No, it was not intentional.

pandafy, 2024-09-20:
So, the expected configuration would be 

```
  config interface 'wwan0'
        option apn 'apn.vodafone.com'
        option device '/sys/devices/platform/ahb/1b000000.usb/usb1/1-1'
        option ifname 'wwan0'
        option iptype 'ipv4v6'
        option lowpower '0'
        option metric '50'
        option mtu '1500'
        option password 'pwd123456'
        option pincode '1234'
        option proto 'modemmanager'
        option signalrate '5'
        option username 'user123'
```

PR #320 (2024-10-16): [fix] Removed device configuration block for modemmanager interface #316

## Checklist

- [x] I have read the [OpenWISP Contributing Guidelines](http://openwisp.io/docs/developer/contributing.html).
- [x] I have manually tested the changes proposed in this pull request.
- [x] I have written new test cases for new code and/or updated existing tests for changes to existing code.
- [x] I have updated the documentation.

## Reference to Existing Issue

Fixes #316
commit 1ddddcff (2024-10-16): [fix] Removed device configuration block for modemmanager interface #316

Fixes #316
closed 2024-10-16 by nemesifier