openwisp/netjsonconfig #335: [bug:openwrt] Creating VLAN 802.1q interface without "network" field

pandafy, 2025-01-09:
**Describe the bug**
Creating VLAN 802.1q field with a blank interface will generating an invalid configuration.

**Steps To Reproduce**

Try generating OpenWrt configuration using the following NetJSON 

```json
{
            "type": "8021q",
            "name": "br-wifi",
            "network": "",
            "disabled": false,
            "vid": 200,
            "ingress_qos_mapping": [],
            "egress_qos_mapping": []
}
```

**Generated Configuration**

```text
config device 'device_'
	option ifname 'br-wifi'
	option name 'br-wifi.200'
	option type '8021q'
	option vid '200'

config interface ''
	option device 'br-wifi.200'
	option enabled '1'
	option proto 'none'
```

shwetd19, 2025-02-10:
Hey @pandafy @nemesifier , I've analyzed the problem and have a proposed solution.

## Issue Analysis

The bug occurs when creating an 802.1q VLAN interface with an empty network field. The current behavior generates an invalid OpenWrt configuration by creating an interface section with an empty name (`config interface ''`), which is not a valid UCI configuration.

## Current Behavior

When the `network` field is an empty string, the converter still creates an interface section but with an empty name, resulting in:

```
config interface ''    # <-- This is invalid
    option device 'br-wifi.200'
    option enabled '1'
    option proto 'none'
```

## Expected Behavior

When the `network` field is empty or not specified, we should only create the device configuration without generating an associated interface section, resulting in:

```
config device 'device_'
    option ifname 'br-wifi'
    option name 'br-wifi.200'
    option type '8021q'
    option vid '200'
```

## Proposed Solution

We should modify the converter to only generate the interface section when the network field has a valid value. Here's a proposed code change:

```python
def render_interface(self, interface, **kwargs):
    network = interface.get('network')
    # Only generate interface section if network field has a value
    if not network:
        return self._render_device_config(interface)
    
    # Rest of the existing interface rendering code
    ...
```

This change ensures we maintain backward compatibility while fixing the invalid configuration generation.

Would you like me to prepare a pull request with this fix? Please let me know if you need any clarification or have questions about the proposed solution.

nemesifier, 2025-02-27:
@shwetd19 yes go ahead, no need to ask for permission.

commit b0def534 (2025-03-19): [fix:openwrt] Auto-generate logical name for VLAN 802.1X interface #335

If the "network" field in the NetJSON is empty, then auto-generate
logical name for the interface.

Fixes #335
commit 5a2513c3 (2025-03-19): [fix:openwrt] Auto-generate logical name for VLAN 802.1X interface #335

If the "network" field in the NetJSON is empty, then auto-generate
logical name for the interface.

Fixes #335

(cherry picked from commit b0def5340ef8da955822646d58c74c3aa11e0e11)
closed 2025-03-19 by nemesifier