openwisp/netjsonconfig #342: error on netjsonconfig: 1.1.1.post1 convert config.tar.gz to json for zerotier

mikysal78, 2025-02-26:
Hi,
I tried uploading the original zerotier config to my router without changing anything.
I saved the backup and ran the conversion instructions.

`netjsonconfig --native config.tar.gz --backend openwrt --method json -a indent="    "`

I get this error:
```
$ netjsonconfig --native backup-ReginaMargherita116-AP-2025-02-26.tar.gz --backend openwrt --method json -a indent="    "
Traceback (most recent call last):
  File "/home/michele/.local/pipx/venvs/netjsonconfig/lib/python3.11/site-packages/netjsonconfig/backends/base/backend.py", line 157, in validate
    Draft4Validator(self.schema, format_checker=draft4_format_checker).validate(
  File "/home/michele/.local/pipx/venvs/netjsonconfig/lib/python3.11/site-packages/jsonschema/validators.py", line 451, in validate
    raise error
jsonschema.exceptions.ValidationError: '0' is not of type 'boolean'

Failed validating 'type' in schema['properties']['zerotier']['items']['properties']['networks']['items']['properties']['allow_global']:
    OrderedDict([('type', 'boolean'),
                 ('title', 'Allow Global'),
                 ('default', False),
                 ('format', 'checkbox'),
                 ('description',
                  'Allow ZeroTier to set Global/Public/Not-Private range '
                  'IPs and Routes')])

On instance['zerotier'][0]['networks'][0]['allow_global']:
    '0'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/michele/.local/bin/netjsonconfig", line 174, in <module>
    instance = backend_class(**options)
               ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/michele/.local/pipx/venvs/netjsonconfig/lib/python3.11/site-packages/netjsonconfig/backends/openwrt/openwrt.py", line 56, in __init__
    super().__init__(config, native, templates, context)
  File "/home/michele/.local/pipx/venvs/netjsonconfig/lib/python3.11/site-packages/netjsonconfig/backends/base/backend.py", line 51, in __init__
    self.parse(native)
  File "/home/michele/.local/pipx/venvs/netjsonconfig/lib/python3.11/site-packages/netjsonconfig/backends/base/backend.py", line 329, in parse
    self.to_netjson()
  File "/home/michele/.local/pipx/venvs/netjsonconfig/lib/python3.11/site-packages/netjsonconfig/backends/base/backend.py", line 348, in to_netjson
    self.validate()
  File "/home/michele/.local/pipx/venvs/netjsonconfig/lib/python3.11/site-packages/netjsonconfig/backends/openwrt/openwrt.py", line 60, in validate
    super().validate()
  File "/home/michele/.local/pipx/venvs/netjsonconfig/lib/python3.11/site-packages/netjsonconfig/backends/base/backend.py", line 161, in validate
    raise ValidationError(e)
netjsonconfig.exceptions.ValidationError: ValidationError '0' is not of type 'boolean'

Failed validating 'type' in schema['properties']['zerotier']['items']['properties']['networks']['items']['properties']['allow_global']:
    OrderedDict([('type', 'boolean'),
                 ('title', 'Allow Global'),
                 ('default', False),
                 ('format', 'checkbox'),
                 ('description',
                  'Allow ZeroTier to set Global/Public/Not-Private range '
                  'IPs and Routes')])

On instance['zerotier'][0]['networks'][0]['allow_global']:
    '0'

```
What am I doing wrong?

I have a need to have a template for the new version of zerotier.
Openwisp even though it has been updated continues to generate the old format.

pandafy, 2025-02-28:
@mikysal78 can you share the your zerotier config (/etc/config/zerotier). Please remove any personal identifiable information (e.g. network id and client secret).

mikysal78, 2025-02-28:
Hi @pandafy,
I tried using the defaults configuration.
I updated openwisp but it doesn't generate the template for the new zerotier format (I don't understand why).
To fix the template I wanted to generate the json code from the original file.

```
config zerotier 'global'
	# Sets whether ZeroTier is enabled or not
	option enabled '0'
	# Sets the ZeroTier listening port (default 9993; set to 0 for random)
	#option port '9993'
	# Client secret (leave blank to generate a secret on first run)
	option secret ''
	# Path of the optional file local.conf (see documentation at
	# https://docs.zerotier.com/config#local-configuration-options)
	#option local_conf_path '/etc/zerotier.conf'
	# Persistent configuration directory (to perform other configurations such
	# as controller mode or moons, etc.)
	#option config_path '/etc/zerotier'
	# Copy the contents of the persistent configuration directory to memory
	# instead of linking it, this avoids writing to flash
	#option copy_config_path '1'

# Network configuration, you can have as many configurations as networks you
# want to join (the network name is optional)
config network 'earth'
	# Identifier of the network you wish to join
	option id '8056c2e21c000001'
	# Sets whether this network is enabled or disabled. If it is not
	# enabled ZeroTier will not join it.
	option enabled '1'
	# Network configuration parameters (all are optional, if not indicated the
	# default values are set, see documentation at
	# https://docs.zerotier.com/config/#network-specific-configuration)
	option allow_managed '1'
	option allow_global '0'
	option allow_default '0'
	option allow_dns '0'

# Example of a second network (unnamed as it is optional)
#config network
#	option id '1234567890123456'
#	option enabled '0'
#	option allow_managed '1'
#	option allow_global '0'
#	option allow_default '0'
#	option allow_dns '0'
```

If you can explain how to generate the new template from openwisp directly I will solve all the problems.
My Openwisp is:

OpenWISP Version: 24.11.1
Installed OpenWISP Modules
openwisp-controller: 1.1.1
openwisp-firmware-upgrader: 1.1.1
openwisp-ipam: 1.1
openwisp-monitoring: 1.1.2
openwisp-network-topology: 1.1.1
openwisp-notifications: 1.1
openwisp-radius: 1.1.1
openwisp-users: 1.1.0
openwisp-utils: 1.1.1
netdiff: 1.1
netjsonconfig: 1.1.1.post1
OS Information
OS version: #1 SMP PREEMPT_DYNAMIC Debian 6.1.128-1 (2025-02-07)
Kernel version: 6.1.0-31-amd64
Hardware platform: x86_64

commit 34166fae (2025-02-28): [fix] Fixed typo in fields of ZeroTier backend #342

Fixes #342
commit 6f05c29f (2025-02-28): [fix] Fixed typo in fields of ZeroTier backend #342

Fixes #342
PR #343 (2025-03-03): [fix] Fixed typo in fields of ZeroTier backend #342

## Checklist

- [x] I have read the [OpenWISP Contributing Guidelines](http://openwisp.io/docs/developer/contributing.html).
- [x] I have manually tested the changes proposed in this pull request.
- [ ] I have written new test cases for new code and/or updated existing tests for changes to existing code.
- [x] I have updated the documentation.

## Reference to Existing Issue

Fixes #342
commit ad5f75fc (2025-03-03): [fix] Fixed typo in fields of ZeroTier backend #342

Fixes #342
closed 2025-03-03 by pandafy